genodelabs/genode #2177: core: handle cap-ref-count overflow

m-stein, 2016-11-23:


m-stein, 2016-11-23:
@nfeske I modified the patch according to your suggestion but I was not able to get rid of the RM session API change. This is because when merely printing a warning on ref overflows in Region_map_component::fault_handler, the ref overflow is triggered again (I assume during the return path in the RPC framework). Throwing an exception instead, fixes this problem. Until now, I was not able to find out where the exception is thrown the second time. When printing out the return pointer in a catch statement in Native_capability, the pointer is bogus. Before start debugging with GDB: Do you have an idea?

nfeske, 2016-11-23:
Unfortunately, I cannot give you an answer from the top of my head.

m-stein, 2021-02-26:
Unfortunately, I'm not sure whether this is still a problem as I was quite lazy with documenting the issue scenario :(

nfeske, 2021-02-26:
Let us keep this issue because it is worth investigating.

nfeske, 2022-05-31:
I'd keep it. You don't need to feel responsible, @m-stein.

m-stein, 2022-05-31:
@nfeske Thanks!