genodelabs/genode #2179: intel_fb_drv: intel_lrc_irq_handler not implemented

chelmuth, 2016-11-24:
In some rare cases of high boot-time concurrency the driver fails to initialize with the following log.
```
[2016-11-21 21:26:46,911] [init -> intel_fb_controller] Warning: no VFS configured
[2016-11-21 21:26:46,911] [init -> intel_fb_controller] --- Intel framebuffer controller ---
[2016-11-21 21:26:46,911] [init -> intel_fb_fs_rom] request for intel_fb_drv -> intel_fb_drv.config
[2016-11-21 21:26:46,911] [init -> intel_fb_drv] --- intel framebuffer driver ---
[2016-11-21 21:26:46,911] [init -> intel_fb_drv] INFO: i2c-core: driver [dummy] registered
[2016-11-21 21:26:47,143] [init -> acpi_drv] Found MADT
[2016-11-21 21:26:47,144] [init -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
[2016-11-21 21:26:47,144] [init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13
[2016-11-21 21:26:47,144] [init -> acpi_drv] Found MCFG
[2016-11-21 21:26:47,144] [init -> acpi_drv] MCFG BASE 0xf0000000 seg 0x0 bus 0x0-0x3f
[2016-11-21 21:26:47,144] [init -> acpi_drv] Found DMAR
[2016-11-21 21:26:47,144] [init -> acpi_drv] 39 bit DMA physical addressable , IRQ remapping supported
[2016-11-21 21:26:47,144] [init -> acpi_drv] DMA remapping structure type=0
[2016-11-21 21:26:47,145] [init -> acpi_drv] DMA remapping structure type=0
[2016-11-21 21:26:47,145] [init -> acpi_drv] DMA remapping structure type=1
[2016-11-21 21:26:47,145] [init -> acpi_drv] DMA remapping structure type=1
[2016-11-21 21:26:47,795] [ 0] DMAR:0xffffffff89323040 FRR:0 FR:0x2 BDF:0:2:0 FI:0x00000000
[2016-11-21 21:26:47,796] [init -> platform_drv -> intel_fb_drv] assignment of 16 succeeded
[2016-11-21 21:26:47,796] [init -> platform_drv -> intel_fb_drv] assignment of 0 succeeded
[2016-11-21 21:26:47,797] [init -> intel_fb_drv] ioremap: mapped phys 0xe0000000 (size 16777216) to virt 0x8000000
[2016-11-21 21:26:47,797] [init -> platform_drv] Error: Device 0x0:0x0.0 is used by more than one driver - session 'intel_fb_drv'.
[2016-11-21 21:26:47,797] [init -> platform_drv] Error: Device 0x0:0x2.0 is used by more than one driver - session 'intel_fb_drv'.
[2016-11-21 21:26:47,798] [init -> platform_drv -> intel_fb_drv] assignment of 248 succeeded
[2016-11-21 21:26:47,798] [init -> intel_fb_drv] Warning: heap could not free memory block
[2016-11-21 21:26:47,798] [init -> intel_fb_drv] [drm:ERROR] blitter: timed out waiting for forcewake ack request.
[2016-11-21 21:26:47,799] [init -> intel_fb_drv] ioremap: return sub range phys 0xe0800000 (size 8388608) to virt 0x8800000
[2016-11-21 21:26:47,799] [init -> intel_fb_drv] INFO: [drm] Memory usable by graphics device = 4096M
[2016-11-21 21:26:47,799] [init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus dpc] registered
[2016-11-21 21:26:47,818] [init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus dpb] registered
[2016-11-21 21:26:47,818] [init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus dpd] registered
[2016-11-21 21:26:47,819] [init -> intel_fb_drv] INFO: [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[2016-11-21 21:26:47,819] [init -> intel_fb_drv] INFO: [drm] Driver supports precise vblank timestamp query.
[2016-11-21 21:26:47,819] [init -> platform_drv] 0:2.0 adjust IRQ as reported by ACPI: 11 -> 16
[2016-11-21 21:26:47,819] [init -> platform_drv] 0:2.0 uses IRQ, vector 0x10, MSI 32bit capable, non-maskable
[2016-11-21 21:26:47,820] [init -> intel_fb_drv] Warning: I/O port(964) write failed
[2016-11-21 21:26:47,820] [init -> intel_fb_drv] Warning: I/O port(965) read failed
[2016-11-21 21:26:47,820] [init -> intel_fb_drv] Warning: I/O port(965) write failed
[2016-11-21 21:26:47,820] [init -> intel_fb_drv] dev_dbg: adapter [DPDDC-A] registered
[2016-11-21 21:26:47,821] [init -> intel_fb_drv] [drm:ERROR] dp aux hw did not signal timeout (has irq: 1)!
[2016-11-21 21:26:47,974] [init -> intel_fb_drv] [drm:ERROR] dp aux hw did not signal timeout (has irq: 1)!
[2016-11-21 21:26:47,977] [init -> intel_fb_drv] [drm:ERROR] dp aux hw did not signal timeout (has irq: 1)!
[2016-11-21 21:26:47,978] [init -> intel_fb_drv] intel_lrc_irq_handler not implemented
[2016-11-21 21:26:48,145] [init -> intel_fb_drv] BUG: failure at genode/repos/dde_linux/src/drivers/framebuffer/intel/dummies.c:611/intel_lrc_irq_handler()!
```

skalk, 2016-11-29:
I have no real solution, as I could not achieve until now to reproduce the interrupt occurrence. However, we do not implement this whole "logical ring" stuff in our intel driver variant. Normally the interrupts are not used at all, except from tiny time frames where the initialization waits for some vblank interrupts. Propably during this period a "logical ring" interrupt occurs, which we do not handle, but stop. I would just implement the "lrc" interrupt handler empty without stopping, and have a look whether this already solves the problem.
@chelmuth what do you think about this?

skalk, 2016-11-29:
commit 2a7e766 can be used to test, whether the initialization still fails or not.

chelmuth, 2016-11-29:
Sounds like a appropriate approach... merged the commit into staging

chelmuth, 2017-02-28:
We experienced no further issues after the dummy implementation was added.