genodelabs/genode #2181: Tweak RAM demand of lxip

chelmuth, 2016-11-29:
Currently, components that make use of lxip require 26-32 MiB of RAM to successfully initialize the library. The high RAM demand partly originates from configuration parameters of the lxip stack, which are left unset in our port. I identified two candidates (but there may be more room from optimization).
* `thash_entries` in *net/ipv4/tcp.c* configuring  the number of hash buckets for TCP connection (currently 524288 8-byte entries) and indirectly the hash table for TCP binds (currently 524288 16-byte entries)
* `uhash_entries` in *net/ipv4/udp.c* configuring the number of hash buckets for UDP/UDP-Lite connections (currently 65536 32-byte entries)

nfeske, 2021-02-26:
Let's archive this issue.

ssumpf, 2023-12-13:
While porting lxip to 6.1.20 I stumbled over big hash table allocations again (up to 16MB for one table). The numbers described by @chelmuth are still the same and concern thash/uhash. Reading https://lwn.net/Articles/906186/ it became clear that one hash entry correlates to one socket and that the default configuration (about half a million sockets for TCP) is made for server loads and a single IP Stack within the kernel. Since in Genode a component has a dedicated IP stack or uses the one of a VFS server, we could severeley limit the number of hash entries per IP Stack. Even if there are not enough hash entries for a specific work load left (hash collisions), buckets will be created.

ssumpf, 2023-12-13:
A test run with the default settings (524288 TCP and 65536 UDP) results in 27.25 MB of hash table allocations. When limiting this to 2048 hash entries for each protocol 2.48 MB are allocated where 2MB are caused by the `modern` routing table which is static. Because these allocations are performed once during initialization there is a lot RAM quota that can be saved here (~25MB). I will test if this setting has any influence on the performance.

ssumpf, 2023-12-13:
68ac134 implements thash/uhash settings for the current lxip stack and could be used for nightly testing. The `2048` is not set in stone and could be made configurable - if required. 36ea3cc demonstrates the possible decrease in memory usage for the nerperf test. I did not experience any performance regressions so far.

closed 2024-02-29 by nfeske