genodelabs/genode #2190: Name components unambiguously

nfeske, 2016-12-03:
To proceed with the unification of build directories across different kernels, we have to overcome the following problem:

Right now, we use the specs mechanism to select a particular version of a component depending on the build directory's configuration. E.g., even though the timer driver is always named "timer", it is a different component depending on the used kernel. For such components, we assume that there is only one version that matches the specs as configured for the build directory. In practice, this is already ambiguous. For example, there exists a PIT-based timer implementation that in principle works on all x86 platforms. On NOVA, however, there is also a NOVA-specific timer implementation that uses a kernel-specific mechanism. Hence, for a given `nova_x86_64` build directory, both timers would match. The same problem also occurs with the framebuffer driver as both the VESA and Intel-framebuffer drivers can be used on Intel-based platforms. In the future when sharing a build directory by multiple kernels, this situation will eventually become the common case rather than an exception.

Thanks to an offline discussion with @m-stein on Friday, I'd like to propose the following way to untangle the situation:

We consistently name components after their actual implementation. E.g., instead of naming all framebuffer drivers "fb_drv", let us call them "imx5_fb_drv", "vesa_fb_drv", "omap4_fb_drv", etc. Instead of naming all timers "timer", let us call them "nova_timer_drv", "pit_timer_drv", "foc_timer_drv", etc.

Of  course, we do not want to put the burden of selecting the appropriate drivers on each run script. Most run scripts don't care about the particular drivers that are used. They just need a "framebuffer" or a "timer" or a "nic". How to bridge this gap? I suggest _a little bit of magic_: In addition to the regular binaries that can be specified as boot modules to the `build_boot_image` function, the function also accepts a set of known generic names, specifically "core", "timer", "framebuffer", "nic", "input", "block", and "ld.lib.so". In reality, none of those binaries exists. But the (already kernel-dependent) boot-dir plugin of the run tool maps those names to actual binaries according to _heuristics_ present in the respective plugin. Note that the boot-image plugin may accept configuration arguments that enable the user to tweak the heuristics, e.g., extending the RUN_OPT with `--timer pit_timer_drv`, one could tell the `boot_dir/nova` plugin to use the PIT-based timer instead of the  default NOVA-based timer.

By following this way, I envision the following situation.

* A build directory can be created for `x86_64`, which is in principle compatible with all kernels that support this architecture.
* In the build directory's `etc/build.conf`, you can add multiple `base-<kernel>` repositories, e.g., `base-nova` and `base-hw`.
* In the build directory's `etc/specs.conf`, you can add kernel-specific spec values to enable the build of NOVA-specific and hw-specific components. Maybe, this manual step could be replaced by just incorporating all specs found in the `<repository>/etc/specs.conf` files of the enabled `REPOSITORIES`.
* The build system works as usual. But in contrast to today where only a single timer would be built when issuing `make drivers/timer`, the new version would build all timer drivers that are compatible with any of the kernels, in the specific case, this would be `pit_timer_drv`, `hw_timer_drv`, and `nova_timer_drv`.
* To execute a run script, one has to specify the kernel that should be used by adding a `--kernel` argument to `RUN_OPT`. The kernel-specific `run/boot_dir` plugin would then pick the most appropriate timer component and integrates it as "timer" into the boot directory. Hence, the init configuration can simply refer to a "timer" exactly like today.
* Run scripts that demand a specific version of a driver can still do so. They would simply not use the magic words but specify the concrete instance of the needed component at both the list of boot modules and the init configuration. So the concept would scale to complex scenarios, e.g., where multiple different NIC drivers are integrated into a single scenario.
* Inside an ARM build directory, the diversity of drivers is even higher. Here, the `run/boot-dir/hw` plugin should accept a `--board` argument. Based on given value (like `rpi`, `armory`, `pbxa9`), it would take the decision about the drivers to use. If a board lacks a driver that is needed by the run script, it could print a meaningful message. The heuristics to select the most suitable drivers per board could possibly be shared by the different kernels that support ARM.

Note that I am aware that this approach is not perfect for everything. For example, it cannot capture the fact that `fb_sdl` is both an "input" and "framebuffer" driver, or that several ARM platforms depend on the USB driver to play the role of "input", "nic", and "block". So we still need to handle those aspects separately.

Is it nevertheless a reasonable approach? I know that "magic" and "heuristics" are universally considered to smell bad. But at some point, we have to supplement those heuristics (which driver to use on which kernel/board). My proposal keeps them largely out of the individual run scripts and puts them in the hands of the boot-dir plugins. I hope that this approach will accommodate the most common use cases for run scripts while being convenient to use. For the latter reason, I would not even introduce a formalism for syntactically marking those magic words as long as the set of magic words is relatively small (like the list I gave above).

ValiValpas, 2016-12-06:
@nfeske Seems reasonable. I'm sure we could provide some ideas about how to approach this in a less heuristic and more formal way. Let me know when you reach the point at which you start supplementing those heuristic.

jklmnn, 2018-08-08:
Two things about this:
I'm not sure if it is good to always build all drivers that could be used in this platform for a specific type of driver. For the timer this might not be a problem, but e.g. building the intel_fb takes a reasonable amount of time and doing so when only VESA is needed increases the build time noticeably.

The second might not fit exactly into this ticket but I think this approach somehow could be extended to build components, especially libraries. To me it is often not clear which shared libs a specific component requires (and searching through the dependencies and which of them are shared is not a really good option) so my way to go is to run and look what is missing. Is there a way to provide this more explicitly?

chelmuth, 2018-08-08:
With the depot archives available your first remark is an obsolete issue IMO. In the future, you will build and use those archives from the depot you need for your scenario and no longer use the `build` directive in your run script. For your example you have to decide for *bin/x86_64/intel_fb_drv* or *bin/x86_64/vesa_drv*.

The information you're seeking with your second remark can easily be derived from the ELF binary's NEEDED sections, for example,
```
build/x86_64> objdump -p bin/fetchurl | grep NEEDED
  NEEDED               curl.lib.so
  NEEDED               ld.lib.so
  NEEDED               libc.lib.so
  NEEDED               libcrypto.lib.so
  NEEDED               libssh.lib.so
  NEEDED               libssl.lib.so
  NEEDED               pthread.lib.so
  NEEDED               vfs.lib.so
  NEEDED               zlib.lib.so
```

chelmuth, 2018-08-08:
If you have to update the base package, depending (i.e., all) depot archives also have to be updated. To alleviate this massive rebuild I use ccache.

skalk, 2019-05-03:
AFAIK the components left are `ahci_drv`, `uart_drv`, `sd_card_drv`, `fb_drv`, `ps2_drv`, `rtc_drv`, and `audio_drv`, whereby the last two are less critical, because they exist for the x86 architecture only anyway.

chelmuth, 2019-05-03:
*rtc_drv* differs between *pc* and *linux*.

skalk, 2019-05-03:
Ok, I additionally solved the issue for the `fb_drv`, see above commits, or the whole commit series at my "merge_to_staging" branch.

skalk, 2019-05-03:
With above commit series all drivers are now named explicitly refering their device resp. the board they belong to. The only exception are some drivers for x86 non-linux drivers, like `ahci_drv`, `rtc_drv`, or `ps2_drv`. Currently, this is a non-issue, because they are unique regarding the x86 architecture, and therefore do not interfer in a common build directory. As we now can use unified build directories also for arm, I see this issue as solved.

tomga, 2019-05-04:
While I understand idea behind this issue I have doubts about one aspect of implementation.

Board specific REQUIRE were replaced to more generic architecture specific versions (e.g. for rpi framebuffer in [target.mk](https://github.com/skalk/genode/blob/e407b714fbc9901bbd8bf7c23e63d0c293a74a48/repos/os/src/drivers/framebuffer/spec/rpi/target.mk) rpi was changed to arm_v6. This driver can be used without any changes for later rpi versions. And they are arm_v7 and later.

Is this REQUIRE needed for anything? Maybe it could be removed completely and is left just because it was easier/safer?

skalk, 2019-05-06:
@tomga I've chosen the REQUIRE conversatively here. But if the driver runs unchanged on rpi2 and rpi3, we can change the REQUIRE to generic arm. In general, the REQUIRE is needed, as long as architectural dependencies exist. But even if there are no such dependencies, REQUIRE is useful when the component cannot run on another architecture anyway, e.g., this driver will never run on x86 or riscv, but without the REQUIRE it would be build for those architectures too, whenever make drivers/framebuffer gets invoked. I'll will change this for the framebuffer driver. If you found other drivers for rpi that can be used without modifications, please let me know.

tomga, 2019-05-06:
I can change that later when I get to integrating my changes. I just wanted to know if I correctly understand situation and it seems that yes.

chelmuth, 2019-07-09:
We finished this work with release 19.05.

closed 2019-07-09 by chelmuth