genodelabs/genode #2193: nic_router.run: make it work on all platforms

m-stein, 2016-12-06:


m-stein, 2016-12-19:
@chelmuth With these commits, the results look as follows:
```
hw_arndale     - http_client_1/2 cannot connect, rest ok
hw_panda       - http_client_1/2 cannot connect, rest ok
hw_rpi         - nic_router doesn't come up
hw_x86_64_muen - build error (not related to nic_router)
foc_x86_32     - fails to read config
nova_x86_32    - fails to read config
okl4_x86       - fails to read config
pistachio_x86  - fails to read config
fiasco_x86     - fails to read config
sel4_x86       - fails to read config
hw_imx53       - not supported
hw_odroid_xu   - not supported
hw_riscv       - not supported
hw_usb_armory  - not supported
hw_wand_quad   - not supported
linux_x86      - not supported
hw_pbxa9       - ok
hw_x86_64      - ok
hw_zynq        - ok
foc_pbxa9      - ok
foc_x86_64     - ok
nova_x86_64    - ok
```
I'll go further into the x86_32 and arndale/panda problems this week. Anyway, it would be nice to have these commits on staging now. They are already a step forward.

m-stein, 2016-12-22:
With the new 4 commits https://github.com/m-stein/genode/commit/3305b2e578242b68d08b4f7a6a2c02053720be5d, https://github.com/m-stein/genode/commit/4a7f9ff23b3194b1fcd47fedb063f8ba37d25fae, https://github.com/m-stein/genode/commit/5f4c7c70aa275d7be1095e864ac40ab7063055e6, https://github.com/m-stein/genode/commit/8e0f2fbce92fce7292d1f35cdb0c8b5acb64d293, Tor and x86-32 work again.

m-stein, 2017-01-19:
@nfeske Commit 0dbf710ea20e19c2b24f5104afd37e3a2af5bb66 circumvents a newer issue in the LWIP cleanup.

m-stein, 2017-02-17:
@nfeske These commits fix some recent problems (higher RAM need, VFS tags, ...) an older one (static IP configs) and show-case the use of the new run script style.

9c511b1 nic_router.run: raise Qemu RAM size
6da0ff0 nic_router.run: do not append but concatenate
4aa3394 run/platform_drv: add append-free interface
7def5a6 lwip/http_clnt: remove unused variable
ec5758b nic_bridge tests: simplify test names
11a4272 nic_router.run: use static IP config in general
92e4b93 nic_router: fix indentation in verbose log
64b89a2 lxip/lwip tests: non-libc attributes to config tag
188cb99 nic_router.run: update to new usage of the VFS tag

nfeske, 2017-02-17:
Thank you for the commit series. I merged all commits to staging.

m-stein, 2017-02-23:
@nfeske Added further commits:
7c83793 nic_router.run: avoid use of 'string cat'
a9b3382 input_filer: fix typo in README

chelmuth, 2017-09-29:
Are we there already?

m-stein, 2017-10-04:
This issue was definitely fixed at some point in time. So, I'll close it although I cannot verify it now due to changes in progress with lwip. For regressions we should open new issues.

closed 2017-10-04 by m-stein