genodelabs/genode #2195: Genode binary application interface (ABI)

nfeske, 2016-12-08:
To make practical use of Genode's kernel-agnostic binaries, we need to cleanly separate Genode's binary application interface (ABI) from the concrete kernel-specific implementation. The ABI is provided by the library interface of the dynamic linker. Until now, each concrete instance of the dynamic linker is called `ld.lib.so` regardless of the kernel. This works because each variant resides in a separate build directory (which is currently tied to one kernel each). In line with the discussion of #2190, I intend to name each variant of the dynamic linker according the the respective kernel that is used to implement the ABI. The common ABI is represented by a `ld.lib.so` stub library that merely contains symbol information but no code.

Components will be linked against the generic ABI. At integration time, the generic ABI stub will be replaced by the concrete instance of the dynamic linker that matches the used kernel. The latter step will be transparently handled by the run tool's boot-dir stage.

chelmuth, 2016-12-20:
As it seems ld.lib.so must provide typeinfo data objects for base types implemented by the supc++ library, we add all those symbols to the ld symbol list. I generated the list by
```
genode-arm-g++ -fPIC c.cc -co c.o | genode-arm-nm --format=posix c.o | grep _ZTI
```
with *c.cc* as follows
``` c++
#define EXPAND(type) \
	catch (type) { } \
	catch (type *) { } \
	catch (type const *) { } \
	catch (type const volatile *) { } \
	catch (type volatile *) { }

extern void f();

void typeinfo_deps()
{
	try { f(); }
	EXPAND(bool)
	EXPAND(char)
	EXPAND(short)
	EXPAND(int)
	EXPAND(long)
	EXPAND(long long)
	EXPAND(signed char)        /* this is not 'char' */
	EXPAND(unsigned char)
	EXPAND(unsigned short)
	EXPAND(unsigned int)
	EXPAND(unsigned long)
	EXPAND(unsigned long long)
	EXPAND(float)
	EXPAND(double)
	EXPAND(long double)
}
```

nfeske, 2017-02-28:
Fixed in Genode 17.02

closed 2017-02-28 by nfeske