genodelabs/genode #2196: IO framework: separate MMIO access from IO logic

m-stein, 2016-12-09:


ValiValpas, 2016-12-12:
Nice one. Based on this, I could now implement a class for accessing I2C slave devices.
In order to use this, a platform-specific I2C driver is required. Such a driver as well as a test accessing the ADV7511 slave available on the parallella board can be found here: https://github.com/ValiValpas/genode-world/commits/zynq_i2c_io

nfeske, 2016-12-12:
As a minor point that I already mentioned to @m-stein today, I would prefer to rename `Io` to `Register_set`, which has a concrete meaning that fits the purpose of the class. To me, it intuitively makes much sense to define `Registers` within a `Register_set` as opposed to `Io`, which is rather cryptic.

chelmuth, 2017-02-28:
What about this contribution? For me it looks @nfeske expressed a minor concern about the naming that was not addressed.

m-stein, 2017-03-01:
@chelmuth @nfeske I rebased the commits and also added the new variadic wait_for method. The commits now preserve the possibility to use the old wait_for methods without modification but mark them deprecated. I only started porting some drivers to the new wait_for to see if it works.

70ba4b2 exynos5 fb_driver: use new Mmio::wait_for
13ea0ec rpi platform driver: use new Mmio::wait_for
1a724b6 mmio: polling for variable amount of IO conditions
80722ba mmio: separate MMIO access and register logic

m-stein, 2017-03-02:
@chelmuth Added the following commits:
c650312 fixup "rpi platform driver: use new Mmio::wait_for"
d5f05f8 rpi sd_card: move to new Mmio::wait_for
097d8ea omap4 sd_card: move to new Mmio::wait_for
c9e89c4 imx sd_card: move to new Mmio::wait_for
1279887 fixup "mmio: polling for variable amount of IO conditions"
c45876e fixup "os: separate MMIO access and register logic"

chelmuth, 2017-03-02:
Thanks, I merged the series into staging.

alex-ab, 2017-03-03:
With the commits the ahci driver dies (libc_ffat.run) and there are many warnings from the register_set code, like:

```
In file included from repos/os/src/drivers/ahci/ata_driver.h:18:0,
                 from repos/os/src/drivers/ahci/ahci.cc:19:
repos/os/src/drivers/ahci/ahci.h: In constructor ‘Port::Port(Genode::Region_map&, Hba&, Platform::Hba&, unsigned int)’:
repos/os/src/drivers/ahci/ahci.h:430:53: warning: ‘bool Genode::Register_set< <template-parameter-1-1> >::wait_for(typename T::Bitfield_base::Compound_reg::access_t, Genode::Register_set< <template-parameter-1-1> >::Delayer&, unsigned int, unsigned int) [with T = Port::Cmd::Cr; PLAIN_ACCESS = Genode::Mmio_plain_access; typename T::Bitfield_base::Compound_reg::access_t = unsigned int]’ is deprecated (declared at /home/alex/genode/genode.devel/repos/base/include/util/register_set.h:782) [-Wdeprecated-declarations]
   if (!wait_for<Cmd::Cr>(0, hba.delayer(), 500, 1000))
...
```

m-stein, 2017-03-03:
Hm, that's strange.
The warnings are normal. They merely mark the old Mmio::wait_for
interface deprecated. But the commits shouldn't change the behavior of
the framework.

I'll have a look at it.
Thanks for the hint!

Martin

m-stein, 2017-03-06:
These commits solve the problem mentioned by @alex-ab and remove a FIXME:
88cd670 mmio: make base address member private
60623ae base-hw mbi: do not copy Mmap MMIO objects
ff27624 ahci: do not copy port MMIO objects
f2ee22b register set: make noncopyable

m-stein, 2017-03-06:
Added 2 more commits:
8bf9e8a fixup "mmio: make base address member private"
b7f92df bootstrap hw: add missing base/registry.cc

m-stein, 2017-03-07:
Added another commit according to builtbot results
508a8ec fixup "base: extend cxa guards to support threads on various priorities"

chelmuth, 2017-03-07:
Assuming you're referring to d0a262271757ba9cd61df9d71698af38400e3e27 I merged it to staging.

m-stein, 2017-03-07:
Oh, sorry, yes :)

chelmuth, 2017-03-15:
@m-stein is this issue fixed by the commits merged into master today or are there open questions?

m-stein, 2017-03-16:
@chelmuth As far as I can see, only this commit by @ValiValpas is missing on world/master:
https://github.com/ValiValpas/genode-world/commit/aa2250643239b07fcfa4a4d142bb4ffa1a5b57b4

chelmuth, 2017-03-16:
@ValiValpas If the change to genode-world is still relevant, I'd like to ask you to open a pull request there.

closed 2017-03-16 by chelmuth