genodelabs/genode #2197: init does not apply routing policy to initial child log sessions

ehmry, 2016-12-11:


nfeske, 2016-12-12:
This is caused by the fact that now - since cfdbccc5c2cdbbeb76c2d20f93e34322c9bb3478 - the parent creates the child's LOG session as part of the child's environment. The always parent obtains all environment sessions from its parent.

I agree that this behavior is unintuitive and I would like to rectify it. However, at the same time, I would like to employ a coherent semantics for all environment sessions, not to handle LOG as a special case. Consequently, I consider subjecting all environment sessions to the routing policy, including PD, RAM, and CPU. This has a good and a bad consequence:

* Good: we could supply child-provided sessions of those low-level services directly to the children of init. For example, right now when using the CPU sampler, we have to add another init in-between the test program and the sampler to be able to route the test program's CPU session to the sampler. When applying the routing to the environment sessions, this is no longer needed. So services like the CPU sampler become much easier to use.

* Bad: with the current built-in policy, init is guaranteed to never perform an RPC call to any of its children, regardless of the configuration. So init can never and up being stuck, even in the presence of misbehaving children. By allowing the routing of environment sessions to child services, init would - when configured this way - issue RPC calls to child-provided PD/CPU/RAM sessions. If the used service does not answer those calls, init suffers. For example, if the CPU sampler crashed, its immediate parent init would get stuck on the attempt to open a CPU session for the test program.

I think the "bad" point is actually not too bad. First, since the session creation is asynchronous now, a CPU service that does not respond to session open/close requests cannot endanger init. This is the most likely error case (i.e., CPU sampler fails to start up because of too scarce resources or a missing route). The bad case can only happen when the CPU sampler has successfully started, successfully responded to a session request, and then crashes. Second, the cases where CPU/RAM/PD sessions are provided by a component other than core are extremely rare and can roughly be subsumed in one of two categories. Either the scenario is for profiling/debugging/experimentation. In this case, getting stuck is not a catastrophe. When in doubt, the scenario can always be executed by the dedicated init instance to shield the important parts of the Genode system from the experimental setup. Or, the other category, the service adds a security function such as ASLR, monitoring, or similar customizations of core's services. Here, the service is considered as trusted. So it is perfectly reasonable to also trust it when running as a child of init.

What do you think? Should we go for the application of the routing policy to all environment session? I would.

nfeske, 2016-12-12:
This issue can be triggered by the demo.run script when starting nitlog from the launchpad and then attempting to start an instance of testnit. The following error will appear in the log:

```
[init -> scout -> launchpad] Warning: testnit: attempt to request cap for unavailable session: service=LOG cid=4 args='label="testnit", ram_quota=8192' state=0 ram_quota=8192
```

ehmry, 2016-12-12:
Yes, I would like to see the old behavior again, but if it contradicts an intuition that init cannot be disrupted by the behavior of it children, perhaps a warning should be logged.

nfeske, 2016-12-12:
The commit 2069b273053a384f2aa26722c6bca5ea1a342e7e implements the idea outlined above.

commit 286009d4 (2016-12-14): base: apply routing policy to environment sessions

This patch changes the child-construction procedure to allow the routing
of environment sessions to arbitrary servers, not only to the parent.
In particular, it restores the ability to route the LOG session of the
child to a LOG service provided by a child of init. In principle, it
becomes possible to also route the immediate child's PD, CPU, and RAM
environment sessions in arbitrary ways, which simplifies scenarios that
intercept those sessions, e.g., the CPU sampler.

Note that the latter ability should be used with great caution because
in [... cut]
commit 0d295f75 (2016-12-23): base: apply routing policy to environment sessions

This patch changes the child-construction procedure to allow the routing
of environment sessions to arbitrary servers, not only to the parent.
In particular, it restores the ability to route the LOG session of the
child to a LOG service provided by a child of init. In principle, it
becomes possible to also route the immediate child's PD, CPU, and RAM
environment sessions in arbitrary ways, which simplifies scenarios that
intercept those sessions, e.g., the CPU sampler.

Note that the latter ability should be used with great caution because
in [... cut]
closed 2016-12-23 by nfeske