genodelabs/genode #2199: Simpify component startup by `removing component_entry_point` lib

nfeske, 2016-12-15:
Right now, our component-startup procedure provides several hooks to cover a large diversity of components:

* Modern Genode components that use the component API,
* Legacy Genode components that still have a `main` function,
* POSIX applications that use libc and have a `main` function,
* Genode components that depend on libc but also provide a Genode service and thereby use the component API, and
* Hybrid Linux/Genode components.

The hooks make the startup arguably more complex than it should be. One particular case is the hook that we provide for the libc to customize the runtime environment of a libc-using program. Such programs tend to use blocking I/O (read, select), which does not fit well with the asynchronous nature of the component API. Under the hood, the libc bridges this semantic gap by the means of user-level scheduling. To implement this, it needs to intercept the startup of the application code.

Long story short, when linking against the libc, `Component::construct` is transparently intercepted by the libc. With this issue, I suggest to drop the "transparently" part and make the interception explicit. In order to use both Genode's component API and libc I/O functions, a component should implement a `Libc::Component::construct` function. The libc would provide the implementation of the regular `Component::construct` function by setting up the context of the application code and finally calling the application-provided `Libc::Component::construct`.

nfeske, 2016-12-15:
As an intermediate step, it would be good to remove the category of "legacy Genode components that still have a main function" by replacing `main` with a `Component::construct`. This way, we can drop the distinction of both cases as currently done by `component_construct.cc`.

chelmuth, 2016-12-16:
To bridge the gap between the libc component approach and POSIX programs, we should implement a *libposix.lib.so* that implements `Libc::Component::construct` and calls `main`. This library is also the right place to handle the environment variables and command-line arguments. So, we'd finally supersede the `config_args` library and provide `<env name="..." value="..."/>` and `<arg value="..."/>` as configuration nodes of `<libc>`.

nfeske, 2017-02-28:
Fixed in Genode 17.02

closed 2017-02-28 by nfeske