genodelabs/genode #2207: kernel/core output if no serial output is available

alex-ab, 2016-12-22:
Sometimes there is simply no serial connection available - you are travelling and using Turmvilla/Heeselicht or simply there is no UART in your machine. 

To get nevertheless the messages from core or the kernel it would be nice if they are exported as ordinary Genode ROM in order to inspect them if required.

This feature is not meant for very early trouble solving. You will need to have already your system running up to some degree so that you can save the output somewhere (RAM) and additionally have some way of exporting the output - wifi/nic/ahci/sd/usb etc.

chelmuth, 2016-12-23:
Nice to see that this topic is still alive though we discussed last in May 2016. I'll have a look at it over the next week.

NobodyIII, 2017-01-19:
Would GRUB's usb_serial driver be useful?

alex-ab, 2017-12-11:
May you please review 0834628efb28a2eb58d619d656d8304d958ade2a, which adds general support to export the messages of the Log core service by a ROM. 

745293a88d663fbb63a09ee3b1f86c02f7e6da15 and a4f7064dd3cb339b1d067f61d5e01836384af16a showcases how it would be enabled per Genode base platform (unify this part ?).

nfeske, 2017-12-11:
@alex-ab Once all platforms are adapted, we can remove the original `init_log` function, right?

alex-ab, 2017-12-11:
I don't know, the init_log is called very early and some comments say something about linux hybrid which I'm not familiar with

nfeske, 2017-12-11:
OK. Then it might be better to call your new function `init_core_log` and move its declaration to a core-only header (not `globals.h`, which is used by all components, not only core).

alex-ab, 2017-12-12:
I incorporated your review commits in the issue [branch](https://github.com/alex-ab/genode/commits/issue_2207_17_12). Should all platforms support the exporting of the core log ? Currently okl4, pistachio and linux (I will do tomorrow) are missing on the topic branch.

nfeske, 2017-12-12:
This looks very good. Thank you for incorporating all my comments! I think skipping the support for the mentioned three platforms is fine.

alex-ab, 2017-12-12:
Ok, updated the [issue branch](https://github.com/alex-ab/genode/commits/issue_2207_17_12) by adding okl4 and pistachio support and incorporated the comments. I also made the test autopilot ready.

alex-ab, 2017-12-12:
Regarding the platforms/kernels, just select what you think is worth supporting and drop the others. (you/we may also squash the selected platforms to one commit)

alex-ab, 2017-12-13:
Does the example commit a9edd5cf75aa9039ea3f75a7157ac88673c66ebe covers the expressed requirement during our sculpt session to disable serial output if not wanted, implemented by fcb214f8296c8ef9cb683106264af6baa3ede55b ?

nfeske, 2017-12-13:
The approach and naming of the hook function `run_output_serial` is fine. However, in Tcl, boolean values are usually expressed as `0` and `1`. The words `true` and `false` are just strings. The `expr` function takes special steps to handle those strings but it is better to stay away from them.

I am a bit afraid that the innocent-looking option to disable the serial output results in a structural change of how the system is booted by skipping bender. I.e., depending on the use of this option, the physical memory layout will look different, potentially triggering issues that are unrelated to the serial log.

nfeske, 2017-12-13:
I wonder, what was the reason to not let bender write a 0 to the BDA when encountering a 'noserial' command-line option? This would certainly be the least invasive approach, wouldn't it?

chelmuth, 2017-12-13:
According to http://wiki.tcl.tk/16235 we're currently using `true` and `false` correctly but I've to admit the disussion in this Wiki leaves an aftertaste.

nfeske, 2017-12-13:
I gave the new `core_log` a try and would like to change one detail. Right now, the component requests two ROMs "core_log" and "kernel_log". If one of those session requests fails, it prints an error and proceeds. I would like to simplify the component such that it opens only one ROM with the name "log". If it fails, it stops. For logging both the kernel log and core log, we would use two instances. This has the following advantages:

* It removes the red error message on platforms that lack the `kernel_log` ROM (all platforms except NOVA).
* It simplifies the component while tying it less to the current specific use case. E.g., it could be used for other sources where data streams are propagated via ROM modules.
* The core and kernel messages can be routed to different LOG services.

nfeske, 2017-12-13:
I see that the third point is already possible with the current version (there are 2 log sessions created).

nfeske, 2017-12-13:
I merged the current version to staging.

alex-ab, 2017-12-13:
> I gave the new core_log a try and would like to change one detail. Right now, the component requests two ROMs "core_log" and "kernel_log". If one of those session requests fails, it prints an error [...]

272f80d354e8fab8af1bef9dd0379244144eeef7 should address it

alex-ab, 2017-12-13:
> I wonder, what was the reason to not let bender write a 0 to the BDA when encountering a 'noserial' command-line option? This would certainly be the least invasive approach, wouldn't it?

There is no such feature in bender 'noserial' - the tool is meant to enable serial.

nfeske, 2017-12-13:
Thanks for the fixup 272f80d354e8fab8af1bef9dd0379244144eeef7. Merged to staging.

nfeske, 2017-12-13:
Fixup 2d23721cbf48f18523c928ad2b1cea2f38aec345 adds a depot recipe for `log_core`.

nfeske, 2017-12-13:
> There is no such feature in bender 'noserial' - the tool is meant to enable serial.

This statement (and your patch) suggests conversely that removing bender effectively disables the serial output. This is unfortunately not true. :-/

nfeske, 2017-12-13:
I added the core log to my `sculpt_next` branch via 6e3710c98fd4ad2568e1d048ba16b9b796235681. It works perfectly. Thanks @alex-ab!

chelmuth, 2017-12-21:
Could we consider this fixed? If there are loose ends please document here.

alex-ab, 2018-06-01:
@chelmuth: please review, I adjusted both commits according to our discussion in the morning

commit f828c625 (2018-06-01): nova: export kernel messages via memory

Fixes #2207
commit 70028f84 (2018-06-12): nova: export kernel messages via memory

Fixes #2207
closed 2018-06-13 by chelmuth