genodelabs/genode #2208: fb_drv: Linker::Not_found

ehmry, 2016-12-22:
The intel framebuffer driver is not coming up for me, unless I change the ``base`` library dependecy to ``base-nova``.

The error:
```
[init -> intel_fb_drv] Error: LD: could not lookup symbol "ips_link_to_i915_driver"
[init -> intel_fb_drv] Error: LD: failed to load program
[init -> intel_fb_drv] Error: Uncaught exception of type 'Linker::Not_found'
[init -> intel_fb_drv] Warning: abort called - thread: ep
```

chelmuth, 2016-12-22:
Which branch/head are you using?

ssumpf, 2016-12-22:
@ehmry: As a quick fix, you could try to add
`void ips_link_to_i915_driver(void)
{
	TRACE_AND_STOP;
}` to repos/src/drivers/framebuffer/intel/dummies.c

ehmry, 2016-12-22:
master

ehmry, 2016-12-22:
@ssumpf ``ips_link_to_i915_driver`` is used by the driver, and should come from a static library.

ssumpf, 2016-12-22:
@ehmry: I know, but it is dynamically linked now which changes things. I had the same issue on my mesa-11 branch where I had to provide additional dummies, definitions, and implementations just because I converted the driver into a shared library. This function is not really called, they just require its address and because it is a weak symbol, the linker gets activated. Have a look at https://github.com/ssumpf/genode/tree/mesa-11 The Intel driver there resides under reops/dde_linux/src/lib/frambuffer/intel Any missing parts should be in dummies.c, lx_emul.h, and lx_emul.cc.

ssumpf, 2016-12-29:
@chelmuth: Please merge  872d621

commit d9a5173a (2016-12-29): intel_fb: Fix dynamic linking issues

Fixes #2208
commit dfae61c7 (2017-01-13): intel_fb: Fix dynamic linking issues

Fixes #2208
closed 2017-01-13 by nfeske