genodelabs/genode #2209: usb_drv problems with dynamic linking

ehmry, 2016-12-22:
Usb_drv no longer announces the Block service for me, and I have bisected the problem to ccffbb0dfce3fa85a152ad24b7c4494109ee5f8e.

I have a boot log here
https://gist.github.com/ehmry/853611e1c786b92acb5c6dfd684e613a

ehmry, 2016-12-22:
I was wrong, I now believe this to be a real hardware problem.

chelmuth, 2016-12-23:
Your log looks strange. Is this an Intel platform with Intel graphics? Also all those *assignment* messages puzzle me
```
[init -> log_lower] [platform_drv -> intel_fb_drv -> ] assignment of PCI device 0:2.0 succeeded
[init -> log_lower] [platform_drv -> intel_fb_drv -> ] assignment of PCI device 0:0.0 succeeded
[init -> log_lower] [platform_drv -> intel_fb_drv -> ] assignment of PCI device 0:3.0 succeeded
[init -> log_lower] [platform_drv -> audio_drv -> ] assignment of PCI device 0:3.0 succeeded
[init -> log_lower] [platform_drv -> nic_drv -> ] assignment of PCI device 0:1.0 succeeded
[init -> log_lower] [platform_drv -> usb_drv -> ] assignment of PCI device 0:1.0 succeeded
[init -> log_lower] [platform_drv -> ahci_drv -> ] assignment of PCI device 0:3.2 succeeded
[init -> log_lower] [platform_drv -> usb_drv -> ] assignment of PCI device 0:2.0 succeeded
```
Could you please check that this log message uses reasonable input values and formatting?

chelmuth, 2016-12-23:
Also the log shows DMAR (IOMMU) faults
```
[ 0] DMAR:0xffffffff810340b0 FRR:0 FR:0x6 BDF:0:1d:0 FI:0x02827000
[ 0] DMAR:0xffffffff810340b0 FRR:0 FR:0x6 BDF:0:1a:0 FI:0x0283d000
```
On x201, PCI 1d.0 and 1a.0 are EHCI USB controllers. That would explain why also simple DMA transactions do not finish.

reet-, 2016-12-23:
@chelmuth In our experience this does not indicate a general DMA failure. Such faults can be caused by EHCI controllers trying to access their reserved memory regions (RMRR) during init (see _Handling Requests to Reserved System Memory_ in the VT-d spec). You can check the ACPI DMAR table for the actual addresses of a platform.

chelmuth, 2016-12-23:
@reet- That's the task of our acpi_drv which logs detected RMRRs found in the DMAR. Nevertheless, this late in boot process the device should not access those RAM regions and the log messages hint some issue. usb_drv also implements the USB handover from BIOS to OS.

reet-, 2016-12-23:
@chelmuth If it is late in the boot process, it is def. strange.

ehmry, 2016-12-23:
That X201 really belongs in retirement, I'm not worried about it.

ehmry, 2016-12-24:
Now I can't use the intel_fb_drv on a different laptop when I have VT-d enabled by the BIOS.

https://gist.github.com/ehmry/a4f7413e88b9d2242c30a7828ae0070f

The plot thickens?

alex-ab, 2016-12-25:
@ehmry: I have seen something like this with 16.11, which should have be gone however after e93ef168a1a66756932e743afa0c9af0f51469b3 on master.

alex-ab, 2017-01-06:
@ehmry: we found the issue - please try to replace in device_pd/target.mk 'base' with 'base-nova' - which helped for me.

However the problem may at any time happen again - to actually solve the issue, I think we can't avoid to re-work the platform_drv <-> device_pd handling structurally.

ehmry, 2017-01-07:
@alex-ab, yes, that fixes it for me, thank you

nfeske, 2017-02-28:
Fixed in Genode 17.02

alex-ab, 2017-02-28:
To quote myself:

> However the problem may at any time happen again - to actually solve the issue, I think we can't avoid to re-work the platform_drv <-> device_pd handling structurally.

Please let open until this gets really fixed. I plan to do it as soon as I find the free time slot.

nfeske, 2017-02-28:
@alex-ab thanks for keeping an eye on my manic issue-closing assault. ;-)

alex-ab, 2017-08-02:
Now had the time to look into. Thx @nfeske for the discussion and pointers.

alex-ab, 2017-08-18:
@chelmuth: please consider for staging eb5bce5f79468b87656d4403a0622e4677c2fa7f

commit e5bbe1cf (2017-08-18): nova: fix implement Pd_session::map

Fixes #2209
closed 2017-08-30 by chelmuth