genodelabs/genode #2210: base-hw/x86_64: Translation_table fails when PML4 is used

senier, 2016-12-26:
In a custom application I'm using the Translation_table implementation from base-hw/x86_64 to manage an 48-bit address space. When inserting a translation that makes use of PML4 (i.e. an address  >= 2^39), the operation fails with page fault.

Example code fragment:

```
   Translation_table *ppgtt;
      if (!gpu_allocator.alloc (sizeof (Translation_table), (void **)&ppgtt)) throw -1;
    ...
    addr_t va = 0xdeadbeef000;
    Genode::log ("Mapping va=", Genode::Hex(va), " to pa=", Genode::Hex((addr_t)scratch_pa),
                 " (base of scratch=", Genode::Hex((addr_t)scratch_addr), ")");
    ppgtt->insert_translation (va, (addr_t)scratch_pa, 4096, scratch_flags, &gpu_allocator);
```

Output:

```
Mapping va=0xdeadbeef000 to pa=0x3382000 (base of scratch=0x7000)
no RM attachment (READ pf_addr=0x23d58 pf_ip=0x100b9a6 from pager_object: pd='init -> intel_fb_drv' thread='ep')
page fault, pd='init -> intel_fb_drv' thread='ep' cpu=0 ip=0x100b9a6 address=0x23d58
```

Apparently, calculating the PML4 table index is wrong, leading to an out-of-bounds access.

skalk, 2017-01-10:
@senier : thank you very much for investigating the problem. We already discussed yesterday in a coffee break that we have to get rid of this manual bit-shifts by using appropriated helper functions to avoid these recurring wrong integer type problem. Unfortunately, we cannot apply a patch without a signed contributor's agreement.

senier, 2017-01-10:
I'll hand one in tomorrow.

commit d8839fdc (2017-03-28): base-hw: Fix PML4 size and constants

The size of the whole table and table entries were exchanged for PML4,
leading to out-of-bound accesses for addesses >= 2^39. Some constants
were lacking type suffixes causing integer overflows.

Added assertions to catch out-of-bound accesses early on.

Fixes #2210
Fixes #2211
commit eb59630c (2017-04-11): base-hw: Fix PML4 size and constants

The size of the whole table and table entries were exchanged for PML4,
leading to out-of-bound accesses for addesses >= 2^39. Some constants
were lacking type suffixes causing integer overflows.

Added assertions to catch out-of-bound accesses early on.

Fixes #2210
Fixes #2211
commit e61a5b0d (2017-05-02): base-hw: Fix PML4 size and constants

The size of the whole table and table entries were exchanged for PML4,
leading to out-of-bound accesses for addesses >= 2^39. Some constants
were lacking type suffixes causing integer overflows.

Added assertions to catch out-of-bound accesses early on.

Fixes #2210
Fixes #2211
closed 2017-05-31 by chelmuth