genodelabs/genode #2214: child: no diagnostic message on missing ROM modules

chelmuth, 2016-12-29:
The child implementation on current master does not print diagnostic messages if ROM modules are missing on child construction. This behavior is by intention as it's now possible to construct children which depend on sessions, e.g., ROM sessions, that will become available later in runtime. Unfortunately, this has unpleasant corner cases. For example, if a run scenario misses ld.lib.so or init, the log output looks like init is stuck during initialization with no messages except core. Higher layers may examine the proposed init status report to reflect why a subsystem is stuck during startup, but for me it's currently not clear how we should proceed at least for the lowest level and with an eye on remote developer support.

nfeske, 2016-12-29:
To diagnose problems at the lowest level, would it appropriate to let core's ROM service print a warning if a non-existing ROM module is requested from core? Alternatively, we could complement init's verbose mode: If running init with `verbose` enabled, it could log whether the child was started immediately or the environment sessions are incomplete right after the construction. I think that for the lowest level, the former case is universally expected.

chelmuth, 2016-12-29:
If it is possible to warn about non-existing ROMs requested by core only, this would definitely help to detect the most basic integration errors. One level up, I'd support to enhance the init verbose warnings as this provides me a tool I can suggest in support cases.

nfeske, 2017-01-06:
The above commit fixes the issue by printing an error message if one of the environment sessions failed immediately, which covers the error case of missing ROM modules at core. Note however, that this fix does not help in situations where the environment session is provided by a child service, which responds asynchronously to the request (sometime after the `Child` construction).

commit db89351e (2017-01-06): init: print error for denied environment sessions

Fixes #2214
closed 2017-01-13 by nfeske