genodelabs/genode #2215: init: services not routing out-of-order

ehmry, 2016-12-29:
I've noticed that services are sometimes not routed and active when annoucements come after session requests.

nfeske, 2016-12-29:
This is strange because the `announce` function of init doesn't do anything other than giving a diagnostic message if the announcement is unexpected: https://github.com/nfeske/genode/blob/pkg/repos/os/include/init/child.h#L790

nfeske, 2016-12-29:
Do you have a reproducible error case? Maybe a possible explanation would be the construction order of the child components. When constructing a child, the environment sessions are routed immediately based on the information that is available for the already existing children. If the service appears after the service-using child, the routing of the environment session may fail. But admittedly, this is just a shot into the blue.

Anyway, when reworking init (I am going to do that next), I will pay attention to this problem. So thanks for reporting it.

ehmry, 2017-01-02:
**run/rom_verify** on my genode-world/issue59 branch triggers this.

nfeske, 2017-01-18:
I started working on this issue today and encountered the following problem:

When applying the routing mechanism to both the environment sessions and sessions that originate from the child, we need a way to distinguish both cases. Imagine a situation where you want to route the environment CPU session to the parent and any CPU session created by the child to the CPU sampler.

The only way to distinguish session requests right now is the label attribute. Unfortunately, the current `label` attribute in the session routes refers to the child's perspective of the session label, not the runtime's perspective. Hence, there is simply no way to route environment sessions with the existing label mechanism.

One way to fix the problem would be to change the meaning of the label attribute to represent the runtime's perspective of the label. This would essentially mean to prefix the existing label values with `child-name -> `. The environment sessions could then be referred to as `child-name` (without the arrow delimiter). I am shuddering because this would break compatibility with all existing configurations and would increase the verbosity of the configurations with very repetitive patterns.

@ehmry @MartijnVerschoor Do you have an opinion or - even better - another idea to address the problem?

nfeske, 2017-01-18:
During a discussion with @chelmuth, we came up with a possible solution.

We'd like to retain the current concise form of `label` arguments, which we consider the common case. For the special case routing an environment session, we introduce a new attribute. Its name should express the meaning of "fully qualified label" but shorter. If specified instead of any of the `label` attributes, the value refers to a label as seen from the perspective of the runtime environment (init). E.g., to define a route for providing access to the dynamic linker used in the environment, one specifies "ld.lib.so". In contrast, to define a route to allow the child to access the dynamic linker binary, one would specify "child-name -> ld.lib.so", or just use the existing label attribute.

We haven't come up with the perfect name for such an attribute. Our current ideas are `strict_label`, `exact_label`, `unscoped_label` (as in contrast to the regular label that lies in the scope of the respective child), or `outer_label`. From these options, I like `unscoped_label` most.

With this feature, we could handle the various cases:

* To route both environment and child-initiated sessions in the same way, one would specify no attribute.
* To define a route that applies to child-initiated sessions only, one would specify one of the attributes `label`, `label_prefix`, or `label_suffix`.
* To define a route that applies only to an environment session, one would specify the new attribute.
* Label attributes cannot be specified together with the new attribute in the same service node.
* For the new attribute, we only need to support the exact match.

nfeske, 2017-01-18:
@ehmry Btw, I found the reason for the "race condition" in your rom-verify scenario. The problem is that your component handles the session-requests ROM manually but does not import the initial state of the ROM. It depends on the ROM session to emit an update signal first. You can fix the problem by calling `handle_session_requests` once in the `Main` constructor.

With my current "init" branch (https://github.com/nfeske/genode/commits/init), your scenario runs reliably for me.

ehmry, 2017-01-19:
Explicitly distinguishing between child metadata and parent metadata sounds practical to me.

I've fixed the original issue in world, so I'm closing this issue.

closed 2017-01-19 by ehmry