genodelabs/genode #2216: ports-foc: remove L4Linux

skalk, 2017-01-03:
Since the very first project that used L4Linux on top of Genode, almost nobody cared much about it. The main developers of Genode do not make use of it, and we do not know of any project currently depending on it. When it comes to OS ABI virtualization, they make use of hardware-assisted VMM like VirtualBox or Seoul. For device driver support, the project also do not follow the L4Linux approach, but drivers are built by ourselfs or we port them using the device-driver environment (DDE) approach.
On the other hand, L4Linux often stands in the way when e.g., modernizing the Genode API like with issue #1987. It also kept me from upgrading to newer Fiasco.OC kernel versions, because of the dependencies of L4Linux with regard to the kernel. In contrast to L4Linux, I know that people are using Genode with Fiasco.OC. I also do so, e.g., when debugging on hardware. When maintaining Genode/Fiasco.OC gets easier by removing the burden of L4Linux I strongly vote for it.

nfeske, 2017-01-03:
Over the course of the past year, I repeatedly had to adjust our version L4Linux to Genode's changes. It definitely required effort to keep it alive. But since it is unused for more than two years (except for our nightly tests) and the kernel version is out of date, the value of this adaptation work seemed questionable. Updating it would wrongly suggest that we are willing to support it, which is not the case either. So I am all for dropping it!

nfeske, 2017-01-20:
The commit https://github.com/genodelabs/genode/commit/cdf7da869cf40b60b1951f8333f9eaf036984929 removes L4Linux.

commit cdf7da86 (2017-01-20): ports-foc: remove L4Linux

Fixes #2216
commit ef5fb30f (2017-01-23): ports-foc: remove L4Linux

Fixes #2216
commit 582e4aec (2017-01-31): ports-foc: remove L4Linux

Fixes #2216
closed 2017-01-31 by nfeske