genodelabs/genode #2217: extend Genode::Env for Libc::Component::Construct

ehmry, 2017-01-03:


ehmry, 2017-01-03:
I am printing a warning when the VFS is configured from ''<libc><vfs/></libc>'' which may not be desireable as this can break tests.

chelmuth, 2017-01-03:
Merged your branch with db85f034b292be3b62e81d935fe08b50501424b3.

ehmry, 2017-01-03:
@chelmuth I've rewritten all the run scenarios I could find, this should make things for intuitive for newcomers.

nfeske, 2017-01-04:
When initially reviewing your patch, I felt a bit uneasy about moving the `<vfs>` config outside the `<libc>` node because in the implementation, the VFS is actually hosted within the libc - a fact that was nicely reflected by the XML structure. Of course, I acknowledge that the new form is easier on the eye and more end-user friendly. A user that configures a component does not care whether the component uses the libc or not.

Now having slept over it, I see that my original line of thinking was a bit flawed. When applying it consequently, it would mean that we need to put the entire configuration of any libc-using component within the `<libc>` node because the entire application code is in fact "hosted" by the libc. This would be super awkward. So my approach was wrong guided to begin with. In fact, technically speaking - as the libc provides the `Component::construct` function, it _is_ the component. Hence, it can naturally evaluate any part of the `<config>` not just the `<libc>` node. Maybe, the introduction of the `<libc>` node was a mistake to begin with? In principle, we could add those attributes into the `<config>` node directly (not a suggestion for immediate action - just a quick thought that crossed my mind).

ehmry, 2017-01-04:
I think that ``<libc/>`` should stay, because to configure stdio anywhere else would seem to imply that it is present regardless of libc.

nfeske, 2017-01-10:
@ehmry I agree with your remark about the `libc` config node.

Looking at the patch above, why do want to expose the heap to the Libc-using application? I would prefer to keep it internal to the libc. In my opinion, a libc-using application should either use malloc/free, or construct a `Genode::Heap` manually. To make the former more convenient, there could be a `Libc_allocator`, which implements the `Genode::Allocator` interface and just uses the malloc/free libc functions.

ehmry, 2017-01-11:
I exposed the allocator just for the sake of reducing redundancy, this is the allocator used to construct the VFS and I think it is otherwise unused. I think you are right that if an allocator is exposed it should not be a different allocator from malloc, so better to remove ``heap()``.

commit db85f034 (2017-01-03): lib/component: pass extended Libc::Env

Libc::Env is the Genode::Env interface extended to cover access
to the XML content of the 'config' ROM and a VFS instance. This
deduplicates the burden of components to attain and manage
these resources.

Fix #2217
Ref #1987
closed 2017-01-13 by nfeske