genodelabs/genode #2230: nova: print debug information for unhandled CPU exceptions

trimpim, 2017-01-10:
We had a division by zero, that only produced a debug output *invalid signal handler*.
Such a message would have saved us hours of debugging.

alex-ab, 2017-01-10:
Maybe you have to elaborate a bit more, since it is supposed to work already. Based on 16.11 release, I get:

```
diff --git a/repos/base-nova/src/test/platform/main.cc b/repos/base-nova/src/test/platform/main.cc
index a7dae58..0f5f498 100644
--- a/repos/base-nova/src/test/platform/main.cc
+++ b/repos/base-nova/src/test/platform/main.cc
@@ -642,10 +642,15 @@ struct Main
        Main(Env &env);
 };
 
+static unsigned long x;
+
 Main::Main(Env &env) : env(env)
 {
        log("testing base-nova platform");
 
+       unsigned long v = 5 / x;
+       log("result ", v);
+
        try {
                Genode::config()->xml_node().attribute("check_pat").value(&check_pat);
        } catch (...) { }

```

produces following output:

```

Genode 16.11 <local changes>
88 MiB RAM assigned to init
Error: ROM session (label="ld.lib.so", ram_quota=4096) unavailable
[init] Error: ROM session (label="ld.lib.so", ram_quota=4096) unavailable
[init -> test-platform] testing base-nova platform
Warning: unresolvable exception 0, pd 'init -> test-platform', thread 'ep', cpu 0, ip=0x10088e9 no signal handler
[init -> test-platform] Error: Uncaught exception of type 'Genode::Ipc_error'
[init -> test-platform] Warning: abort called - thread: main
^CMakefile:293: recipe for target 'run/platform' failed

```

chelmuth, 2017-01-10:
The output is exactly what is needed to fix such errors in low-level components. Unfortunately, the used code base was older than 16.11 as far as I know.

closed 2017-01-10 by chelmuth

alex-ab, 2017-01-10:
I could re-produce the potential issue. See below for a output and the example to trigger it.

In the example below a exception signal handler is registered for a thread, but destroyed without de-registering it explicitly. 

```
Genode 16.11 <local changes>
88 MiB RAM assigned to init
Error: ROM session (label="ld.lib.so", ram_quota=4096) unavailable
[init] Error: ROM session (label="ld.lib.so", ram_quota=4096) unavailable
[init -> test-platform] testing base-nova platform
Warning: submitting signal failed - error 4 - context=cap<37251>
Makefile:293: recipe for target 'run/platform' failed
```

```
diff --git a/repos/base-nova/src/test/platform/main.cc b/repos/base-nova/src/test/platform/main.cc
index a7dae58..e9255c8 100644
--- a/repos/base-nova/src/test/platform/main.cc
+++ b/repos/base-nova/src/test/platform/main.cc
@@ -640,12 +640,28 @@ struct Main
        Genode::Heap  heap { env.ram(), env.rm() };
 
        Main(Env &env);
+
+       void _handle_exception() {
+               error ("exception occured");
+       }
 };
 
+#include <cpu_thread/client.h>
+static unsigned long x;
+
 Main::Main(Env &env) : env(env)
 {
        log("testing base-nova platform");
 
+       {
+               Signal_handler<Main> exception_signal (env.ep(), *this, &Main::_handle_exception);
+
+               Genode::Cpu_thread_client(Thread::myself()->cap()).exception_sigh(exception_signal);
+       }
+
+       unsigned long v = 5 / x;
+       log("result ", v);
+
        try {
                Genode::config()->xml_node().attribute("check_pat").value(&check_pat);
        } catch (...) { }
```