genodelabs/genode #2232: reporter refactor

ehmry, 2017-01-12:
Reporter needs access to an **Env** to make its Report sessions, I'm making this an issue separate from #1987 because it is not so trivial.

nfeske, 2017-01-12:
I already created an interim solution at staging. See b40ca904a1794cca93f1ebde11dd0cb3eca701e4. In short, I added a new constructor that complies with our API and marked the original constructor as deprecated. So we can update all reporter-using components step by step.

ehmry, 2017-01-12:
I noticed your comment a bit late then, but the changes I made in 805578684f3fcb64e177c23563ca3285e3734823 are compatible with the Reporter constructors at b40ca904a1794cca93f1ebde11dd0cb3eca701e4, so a breaking change there can be avoided.

nfeske, 2017-01-12:
Fantastic! May you update the commit message, telling that the patch adjusts components to the changed `Reporter` interface, and rebase the commit to staging? Currently, it does not apply.

ehmry, 2017-01-12:
Corrected at ac7ede2e1199498e6ce6aa02ef1e1dadebf99a68

nfeske, 2017-02-07:
Thanks a lot! Merged to staging.

commit d20f744a (2017-02-07): update Reporter constructors

The Reporter utility needs a reference to the environment at
construction to establish Report connections.

Ref #1987
Fix #2232
commit 48150a70 (2017-02-07): update Reporter constructors

The Reporter utility needs a reference to the environment at
construction to establish Report connections.

Ref #1987
Fix #2232
closed 2017-02-28 by chelmuth