genodelabs/genode #2233: init: revert possibility to route LOG service to child server

MartijnVerschoor, 2017-01-12:
On 16.05 it is possible to route a component's LOG session to an alternative LOG service provided by a child server. In 16.11 all LOG sessions are routed to the parent (core), regardless of configured policies.

This policy is hard-coded in [https://github.com/genodelabs/genode/blob/master/repos/os/include/init/child.h#L653](init/child.h). The same holds for RAM, PD, and CPU. With this implementation it is impossible to provide and route alternative services with these core service names. I’m not sure about RAM, PD and CPU, but for LOG I believe there is a very legit use case to provide, and route to, alternative log-servers.

As a side remark, it would be nice to produce a warning if a configured policy is, not in line with, and ultimately overridden by a hard coded policy.

nfeske, 2017-01-12:
This issue is related to https://github.com/genodelabs/genode/issues/2197.

In fact, I addressed this issue already but apparently, my fix is not complete. I.e., https://github.com/genodelabs/genode/issues/2215 is still open. Also, the hardcoded routes in init stand in the way and should ultimately be removed.

I am planning to revise init in several respects and create a systematic test for it. But I wish to wrap up the current API adaptation work before (there are still some hundred "deprecated" warnings to fix). I expect that the removal of the hard-coded env session routes will break several tests. Hence, I don't want to intertwine this step with the current staging activity. Martijn, I would appreciate you going forward with removing the hardcoded routes from init and report how it works.

MartijnVerschoor, 2017-01-13:
Thanks for the clarification.

Unfortunately removing the hard coded policy is not as simple as commenting [line 653 in init/child.h](https://github.com/genodelabs/genode/blob/master/repos/os/include/init/child.h#L653). In the revised init (16.11), the ‘environment sessions’ of new children are setup immediately during the construction of the Init::Child objects while linearly parsing <start> nodes. Simply commenting out the hard coded policies cause the non-parent provided environment session to be unresolved if the server <start> node that provides it succeeds the child <start> node with a routing policy to that service.

The commit attached to #2215 addresses this issue but - as mentioned - it is not yet a stable solution.

This issue now breaks updating a product to 16.11 that hinges on this functionality, but admittedly I prefer not to step in during this intermediate state of development given the relative complexity of the topic and the short durability of a temporary solution. Do you have a rough time line for when you can continue your revision of init?

nfeske, 2017-01-13:
Thanks for openly communicating your concerns. I ultimately have to weigh the progress of our API adaptation against the inclusion of new features and improvements.

We have made substantial progress (the cleanup work entails over 170 commits so far). But still, many warnings about the use of deprecated API functions remain. However, instead of blocking any feature changes, I'll go ahead an merge the current intermediate state to master (the tests are fine). The only downside are the warnings that may leave a bad impression of users of the master branch.

I will then move on to my work on init, addressing the remaining warnings in a successive way.

MartijnVerschoor, 2017-01-13:
Thanks for your consideration to prioritise, but please note that I only asked for a rough time line.

nfeske, 2017-01-13:
The new init is planned for version 17.02, which will be released end of February.

MartijnVerschoor, 2017-01-13:
Ok, we'll defer finalising our update until then.

nfeske, 2017-02-28:
Fixed in Genode 17.02

closed 2017-02-28 by nfeske