genodelabs/genode #2234: mkisofs not available on Debian

senier, 2017-01-12:
When building Genode on Debian (testing, in my case) the build bails out with

> creating ISO image...
make[1]: mkisofs: Command not found
make[1]: *** [iso] Error 127
Error: ISO image creation failed

Unfortunately, mkisofs is hardcoded and Debian does not provide a compatibility package mapping to genisoimage.

chelmuth, 2017-01-13:
Thanks for your patch which is related to #627 and represents the most pragmatic answer for Debian users. I did not know that Ubuntu and Debian differ in this regard because I tested
```
> dpkg -S $(which mkisofs)
genisoimage: /usr/bin/mkisofs
> lsb_release -d
Description:	Ubuntu 16.04.1 LTS
```
Do you think we should add an error message if the tool is not installed like I did with `check_tool` in f0b75257e180af0efe020de28819d24353a46758 on a devel branch?

BTW your commit is not linked to your GitHub account maybe because of different email addresses in commit and account.

senier, 2017-01-16:
No idea why Debian does not provide a simple symlink like Ubuntu (at least not on testing):

> $ dpkg -S mkisofs
> bash-completion: /usr/share/bash-completion/completions/mkisofs
> $ lsb_release -d
> Description:    Debian GNU/Linux testing (stretch)

Checking for the presence of the tool IMHO is a good idea.

(I added my mail address set in user.email to Github - thanks for the hint!)

commit 263d423f (2017-03-28): tool/create_iso: Make ISO creation tool configurable

Debian has dropped mkisofs long time ago, hence hardcoding it prevents
Genode from being built on Debian easily. This patch makes the tool
configurable (defaulting to mkisofs). Building on Debian can then be
done by something like:

ISOTOOL=genisoimage make run/foobar

Fixes #2234
Fixes #2235
commit 35ed40d1 (2017-04-11): tool/create_iso: Make ISO creation tool configurable

Debian has dropped mkisofs long time ago, hence hardcoding it prevents
Genode from being built on Debian easily. This patch makes the tool
configurable (defaulting to mkisofs). Building on Debian can then be
done by something like:

ISOTOOL=genisoimage make run/foobar

Fixes #2234
Fixes #2235
commit 7ac46dd6 (2017-05-02): tool/create_iso: Make ISO creation tool configurable

Debian has dropped mkisofs long time ago, hence hardcoding it prevents
Genode from being built on Debian easily. This patch makes the tool
configurable (defaulting to mkisofs). Building on Debian can then be
done by something like:

ISOTOOL=genisoimage make run/foobar

Fixes #2234
Fixes #2235
closed 2017-05-31 by chelmuth