genodelabs/genode #2236: Support setting of environment variables in posix lib

chelmuth, 2017-01-13:
With #2218 fixed the posix library handles the parsing of `<arg>` configuration nodes to setup `argc,argv` for posix programs with main routines. As already mentioned in #2199 I suggest the posix library to also be the right place to setup environment variables with `<env name="..." value="..."/>` nodes.

nfeske, 2017-01-19:
I agree that this is the perfect place. But please make sure that it does not collide with noux' existing mechanism for propagating args and env variables to forked and evecve'd processes. Of course, we should eventually update noux to use the new mechanism but we need a migration path (keeping the original mechanism intact for a while).

At the initialization time of the libc_noux plugin, the plugin re-assigns the `genode_envp` pointer to let it point to the noux-prepared environment. As long as this global pointer exists and is used as the backend of the libc env functions, noux should be fine.

ehmry, 2017-01-24:
This was easier than I expected, just populate a list at **environ** (and **genode_envp**) and works like you would expect in libc. Fixed at 8cb7926170049a7e0f9a1f1b2af0c0404076a5cd.

chelmuth, 2017-01-26:
@ehmry would you mind to extend the libc test by some lines for args and env?

commit 8135f1a8 (2017-01-26): lib/posix: populate environment variables from config

Parse ``<env key="..." value=".."/>`` nodes from the config ROM and
populate a list at the 'genode_envp' and 'environ' symbols.

Test script at run/libc_getenv.

Fix #2236
closed 2017-01-31 by nfeske