genodelabs/genode #2238: Fiasco.OC syscall library won't compile anymore

NobodyIII, 2017-01-14:
While I was testing the new unified build directories feature, I found that the syscall library for Fiasco.OC (syscall-foc) won't compile on x86_64 or x86_32. It gives the following error:

genode-x86-gcc: error: syscalls_direct.S: No such file or directory

I later decided to try compiling it for the Raspberry Pi (arm), and I found that it still failed to build, and with a similar error:

genode-arm-gcc: error: atomic_ops_s.S: No such file or directory

chelmuth, 2017-01-16:
Our nightly builds did not show this issue. What commit hash do you use? Did you prepare the foc port before building?

NobodyIII, 2017-01-16:
Oh, it looks like I just needed to install gcc-multilib. That should be mentioned in the list of dependencies for compiling Genode.

chelmuth, 2017-01-16:
I was not aware of this dependency. Could you please explain what causes this? Is it an issue in the Fiasco.OC port (maybe not generating those files)?

NobodyIII, 2017-01-16:
I'm not precisely sure what's going on, but Fiasco.OC's host tools seem to
depend on the Linux system headers. At least on Debian x86_64, without
gcc-multilib, I get an error about asm/errno.h not existing. This leads to
further problems later during compilation.

cnuke, 2017-01-16:
I am running Debian x86_64 with gcc-4.9 without multilib as system compiler and building foc for x86_64 as well as x86_32 with the current master branch and a freshly created build directories works fine. Without more detailed information about your build environment it is hard to tell why it does not work for you.

nfeske, 2017-02-28:
May we close this issue?

chelmuth, 2017-03-15:
No response from the original reporter.

closed 2017-03-15 by chelmuth