genodelabs/genode #2242: UEFI x86_32/64 support

alex-ab, 2017-01-18:
will require another boot-loader than the currently used GRUB legacy. Either we have to develop something own special or go with GRUB2, which seems to be the natural choice.

For the moment I experimented a bit with GRUB2 and that are the currents findings with loading Genode/NOVA and Genode/hw via GRUB2 UEFI.

Serial/UART output
--------------------------
Up to now we used Bender as a chained bootloader to lookup the PCI devices which provide a Serial/UART interface. Bender wrote the findings about the I/O ports into the Extended BIOS Data Area (EBDA), which the kernels and Genode/core looked up to print messages via the UART.

The question is, whether we can keep this way for UEFI (EBDA is not anymore) or we have to find another way, e.g.
- adding a Multiboot or Multiboot2 element about the I/O ports  (non-complex, convenient, not MULTIBOOT/MULTIBOOT2 standard)
- or we expect that the kernel or Genode/Core have to lookup the PCI devices by itself (complex but convenient)
- or if they have to be configured manually (in-convenient).

For the moment I hardcoded the I/O ports in the kernel and Genode/core, which is really in-convenient. But even more in-convenient is, that the PCI serial card for AMT SOL is by default not usable on my testmachine. One has to manually enable in the Command Register of the PCI device the I/O Space bit (http://wiki.osdev.org/PCI) for each boot - grub2 terminal
```
setpci COMMANDS
setpci -s xx:xx.x 4.b=1
```

ACPI
----
The lookup of the ACPI RSDP (Root System Description Pointer) the old way seems to be not possible anymore. The pointer was located either within the first 1KB of the EBDA or in the main BIOS area 0xE0000 to 0xFFFFF.
In the UEFI case the pointer gets reported via the EFI_SYSTEM_TABLE to the bootloader, which is in our case GRUB2.

Unfortunately, GRUB2 does not report the RSDP pointer via the Multiboot structure, however does so for the Multiboot2 structure.

For that reason I extended NOVA to be able to be loaded as Multiboot2 compliant kernel and to parse the Multiboot2 structure. With this information I was able to boot all CPUs and to start Genode/core. Genode/hw booted as normal Multiboot compliant kernel out-of-the-box, since it does not require (currently) the RSDP ACPI pointer.

The next issue is that our platform driver also searches the old way for the ACPI RSDP pointer, which it does not find. In general we will need a way to report the ACPI RSDP pointer via the kernel to Genode/core and from there to the platform driver of Genode. Maybe this is a good point to abanonded the usage of HIP, KIP etc as Genode ROM and instead provide some standardized Genode ROM with absolutely necessary information about the platform which can't be gathered otherwise.

NOVA
-----
With the kernel patch the microkernel can be booted as Multiboot as also Multiboot2.
However an issue shows up on real hardware sometimes. If the GRUB2 bootloader decides to put the Multiboot2 data structure at physical address 0x1000, the NOVA kernel will override it silently and later on the bootstrap of the CPUs will get stuck. One can change the location in the kernel in start.S and lapic.cpp and than everything comes up, but this is a hack and blows up the next time the bootloader chooses the same location as the kernel does.

Further open points
-------------------
- PS/2 still available or replacement required ?
- VESA driver still possible or need some UEFI Framebuffer driver ?

Current state
-------------
I compiled a GRUB2 UEFI 32bit and 64bit version and added it to our tool/boot directory. Additionally I extended our ISO creation tool (mainly use mkisofs and isohybrid the right way) by the both GRUB2 UEFI boot loaders, so that this ISO comes up on UEFI machines.  From my perspective it is reaaally convenient for external users to have one image that boots nearly everywhere, but the dark magic to setup such a hybrid ISO is scary.

Qemu ships not by default with a UEFI firmware, therefore I compiled one of Tianocore/OVMF http://www.tianocore.org/ovmf and added it to tool/boot. 
```
qemu-system-x86_64 --bios tool/boot/tianocore -net none format=raw,file=build/../*.iso -serial stdio
```
or
```
qemu-system-x86_64 --bios tool/boot/tianocore -net none -cdrom build/../*.iso -serial stdio
```

On Qemu I got good first results and it is a good starting point to make further steps. On real hardware, however, the current state is of limited use - because of the open points: manual PCI serial card enablement, manual Serial I/O ports configuration, missing ACPI information in the platform driver and a GRUB2<->NOVA kernel interaction issue regarding the second physical page.

skalk, 2017-01-18:
I know it does not help at all, but I have to express my admiration for this excellent issue description. It reads like a book. Thank you.

nfeske, 2017-01-18:
Thanks for having taken the time to document your findings so well!

alex-ab, 2017-01-19:
Also added a run/image/uefi version which creates a GPT based image which also can be dd to a usb stick (it is simpler/better understandable - means no mkisofs/isohybrid magic, but of course can't be burned as CD/DVD). Thanks @nfeske for the hints regarding mtools, so no sudo required for that to create.

alex-ab, 2017-01-31:
> The next issue is that our platform driver also searches the old way for the ACPI RSDP pointer, which it does not find. In general we will need a way to report the ACPI RSDP pointer via the kernel to Genode/core and from there to the platform driver of Genode.

This part got now fixed. The platform driver as also the acpica application looks into the ROM provided by core which contains now the rsdt/xsdt physical addresses.

alex-ab, 2017-02-02:
> However an issue shows up on real hardware sometimes. If the GRUB2 bootloader decides to put the Multiboot2 data structure at physical address 0x1000, the NOVA kernel will override it silently and [...]

This should be fixed now. I added a segment which comprise the physical range 0x1000, so that the bootloader should be able to put nothing there (the MBI2 structure). Thanks @chelmuth for the hint.

```
Elf file type is EXEC (Executable file)
Entry point 0x201e46
There are 3 program headers, starting at offset 64

Program Headers:
  Type           Offset             VirtAddr           PhysAddr
                 FileSiz            MemSiz              Flags  Align
  LOAD           0x0000000000000100 0x0000000000200000 0x0000000000200000
                 0x0000000000002010 0x0000000000002010  RWE    40
  LOAD           0x0000000000002140 0xffffffff81000000 0x0000000001000000
                 0x0000000000022140 0x0000000002400000  RWE    40
  LOAD           0x0000000000024280 0x0000000000001000 0x0000000000001000
                 0x0000000000000000 0x0000000000001000  RWE    1

 Section to Segment mapping:
  Segment Sections...
   00     .init 
   01     .text .rodata .fixup .init_array .data .bss 
   02     .cpu_ap 
```

alex-ab, 2017-02-03:
> For the moment I hardcoded the I/O ports in the kernel and Genode/core, which is really in-convenient. But even more in-convenient is, that the PCI serial card for AMT SOL is by default not usable [...]

I extended bender to enable the I/O space bit if it is not set - so no manual grub2 intervention required, if bender is used.

chelmuth, 2017-02-03:
Nice idea, and the workaround is anything but misplaced.

alex-ab, 2017-02-07:
> Up to now we used Bender as a chained bootloader to lookup the PCI devices which provide a Serial/UART interface. 

Bender is now, as the NOVA kernel, a hybrid Multiboot/Multiboot2 bootloader. I added several memory checks/overlap detection (for MBI and MBI2) of regions which have been missing (which explains now the strange behavior of kernel as also core when something got overridden silently). Just as a side node, Bender and seL4 kernel overlapped regarding the PHDRS in the ELF partially. 

For the moment I let Bender still write to the EBDA (first physical page) area about the serial port findings, which could be dangerous in UEFI mode (if there is already something). Beside that Bender seems to work seamlessly for me (Qemu and T460p). We should consider to update Bender as soon as possible because of seL4.

chelmuth, 2017-02-07:
Is there already a commit with the new bender version I could cherry-pick?

alex-ab, 2017-02-07:
> PS/2 still available or replacement required ?

Works well on Qemu/UEFI as also T460p. USB HID is also fine.

> VESA driver still possible or need some UEFI Framebuffer driver ?

Intel_fb is working on T460p  + external connector output (ok, some DMAR faults also show up ;-) )
VESA driver means it found a VESA Bios on Qemu (but hangs busy according to the fan of the cpu) and on T460p complains about IOMEM requests which are in the physical RAM allocator - so VESA is not working up to now.

alex-ab, 2017-02-07:
> Is there already a commit with the new bender version I could cherry-pick?

It was part of "enable bender for UEFI (nova & hw)", but I split up in two commits. 
b98de7949ae133113a3edd386704fef3243bf1b1 contains just the binary.

alex-ab, 2017-06-07:
> VESA driver still possible or need some UEFI Framebuffer driver ?

We have now, thx to @jklmnn, a framebuffer running also when booted in uefi mode. I updated this issue to 17.05 and incorporated @jklmnn commits to issue_2242_17_05_fb. At least in Qemu/UEFI it works for me. @jklmnn: please check whether it still works for you on native hardware.

From my point of view we should consider to review the commits and merge.

jklmnn, 2017-06-07:
I have tested it and it still works on the hardware.

alex-ab, 2017-06-09:
The fb driver for mbi2/uefi looks now good, some minor things must be still addressed:
- copyright header are missing
- adherence to Genode coding style ('tool/beautify <filename>' helps here)
- move/rename driver, adjust platform_info ROM

Regarding the last point: The driver does now contain nothing mbi2 or uefi specific anymore.  As long as core provides the ROM with the info of the I/O mem of the framebuffer, width/height, bpp and so on, the driver would also run on other architectures (arm e.g.) or could also run on other bootloaders, as long as they provide the fb information to core and already switched to the specified graphic mode. Because of that, I would recommend to rename the driver to fb_boot_drv and move the driver from spec/* to repos/os/src/drivers/boot (e.g.), since the driver is independent of any SPEC. Other ideas ?

@jklmnn: If you want to address some of the mentioned points, I would be thankful. Otherwise I will do.

alex-ab, 2017-06-09:
Sorry, I meant: 'repos/os/src/drivers/framebuffer/boot'

jklmnn, 2017-06-09:
I can add the copyright headers and adapt the coding style. I hope Monday is early enough.
About the naming, I'm not sure about that but your current idea looks good to me.

alex-ab, 2017-06-09:
Sounds good, then let me do the 3rd point when you're done.

jklmnn, 2017-06-12:
I added the copyright headers and changed the code according ot the beautify tool. I have copied another copyright header and adapted it to my files, I hole this is okay.
You can review and take the commit from https://github.com/jklmnn/genode/commit/cdfd4bcaa664fdaa9da1473dfc35c862b085ab48.

chelmuth, 2017-06-12:
Looking at your license header it seems your code is provided under an MIT-alike license. On the other hand, the driver is heavily using Genode which is published under the terms of GNU AGPLv3 and (in my opinion) is not of much use without Genode. Therefore, I'd like to know if you would mind to change the license to the Genode license. This would not prohibit you as the original author to republish the code under any license you like, but renders the Genode code base more consistent without a mix of per-file licenses. The following license comment would be perfect.
```
/*
 * Copyright (C) 2017 Genode Labs GmbH
 *
 * This file is part of the Genode OS framework, which is distributed
 * under the terms of the GNU Affero General Public License version 3.
 */
```
Also, I have some small remarks to cdfd4bcaa664fdaa9da1473dfc35c862b085ab48 that should be addressed before a merge.

- The patch still refers to Multiboot2/mbi in many places.
- Our coding style prefers include guards over `#pragma once`.
- I'd love a small README that describes the syntax of the ROM information used for the driver. Maybe this is homework we have to do in base, @alex-ab ?
- Finally, I'd like to ask you to merge all commits addressing *repos/os/src/drivers/framebuffer/boot* into one feature commit.

jklmnn, 2017-06-12:
First I'm totally fine with using the AGPLv3, so that's no problem.
I remove the references and in this step I already moved it to `repos/os/src/drivers/framebuffer/boot/`. I have also replaced `#pragma once` with include guards and squashed it into https://github.com/jklmnn/genode/commit/6761683a9d92058b57ac70e361601b19270a9620.

The rom currently has the following structure (appended to the `acpi` tag):
```
<boot framebuffer="true">
    <framebuffer width="1024" height="600" type="1" bpp="32">
        <address>0x40000000</address>
    </framebuffer>
</boot>
```
The `framebuffer` tag is only available if the `framebuffer` attribute is true. Width and height are in pixels, bpp is the depth in bits per pixel and the type should always be `1` (according to the specification there are other types but I have not seen yet such a device).

alex-ab, 2017-06-12:
Thx @jklmnn. @chelmuth: I reworked the commits, according to our discussion.

The [2242 staging](https://github.com/alex-ab/genode/tree/issue_2242_staging) branch contains all relevant commits for UEFI with GRUB2 on NOVA. A new run tool option "image/uefi" was added, which generates a raw disk image containing a GPT partition table with a FAT parition with uefi grub2 images recognized by the UEFI firmware.

The [2242 staging iso](https://github.com/alex-ab/genode/tree/issue_2242_staging_iso) branch contains also the extension for the iso tool to boot via uefi/grub2, which currently is not considered to be merged to staging.

Please review.

jklmnn, 2017-06-13:
Looks good to me. I have only tested the iso branch on hardware now. I have only one question. Is this the expected outcome of all colors mixed?

[image]

I also tested the demo (I switched `fb_drv` to `fb_boot_drv`) and it worked fine and looked correctly.

chelmuth, 2017-06-13:
I hope you don't mind my nitpicking but I'd like to ask you to document and change the `platform_info` ROM for consistency. What the current branch produces looks like follows (Qemu/Tiancore).
```
[init -> acpi_drv] <platform_info>
[init -> acpi_drv] 	<acpi revision="2">
[init -> acpi_drv] 		<rsdt>0x1fe93074</rsdt>
[init -> acpi_drv] 		<xsdt>0x1fe930e8</xsdt>
[init -> acpi_drv] 	</acpi>
[init -> acpi_drv] 	<boot>
[init -> acpi_drv] 		<framebuffer phys_addr="2147483648" width="800" height="600" type="1" bpp="32"/>
[init -> acpi_drv] 	</boot>
[init -> acpi_drv] </platform_info>
```
The `<boot>` and `<framebuffer>` nodes almost meet my expectations. Only the redundant `type` attribute has to be removed and I could also live with just `phys` instead of `phys_addr`. We could also use hex values in the `phys` attribute if you want by generating a string attribute value.

The `<acpi>` looks rather different and also suffers from the use of node content aka CDATA. IMO it is better to just use an attribute value with the (hex) address, e.g.,
```
<acpi revision="2" rsdt_phys="0x1fe93074" xsdt_phys="0x1fe930e8"/>
```
A reason to use nodes could be if we expect more than one occurrence of rsdt/xsdt, but that I don't see. Nevertheless, `platform_info` needs a *README* at a more or less prominent place (not only in the framebuffer driver), otherwise the content structure is only documented by the code using `Xml_generator`.

jklmnn, 2017-06-13:
That sounds logical to me.
@alex-ab I also think we should put a hex value into the `phys_addr` field since it is a memory address. Also I think it would be more readable.

alex-ab, 2017-06-13:
@chelmuth, @jklmnn: I updated the branches according to your comments. Please re-check.

chelmuth, 2017-07-05:
Could we consider this issue as fixed or do I miss something? If there are still open points I'd appreciate a short list here. Thanks.

alex-ab, 2017-07-05:
- support to boot base-hw on UEFI (other kernels ... sel4/foc ?)
- decision whether to take uefi support into iso image or not
- (abandon grub1 over grub2, but maybe this is for another issue)
- (a test case to run nightly)

chelmuth, 2017-07-05:
I'd drop ISO support for UEFI as long as USB mass storage works natively.

ehmry, 2017-08-07:
I've noticed on my T420 I only get 4G of RAM when booting with UEFI but 8G when booting with an ISO image.

For the moment I will check if my firmware or GRUB options are causing this.

alex-ab, 2017-08-11:
Genode/hw is now also able to boot up on UEFI via grub2 with multiboot2 support (qemu for now).
@skalk, @m-stein: may you two please check whether the propagation of the acpi information through bootstrap and core are ok for you or where you would like to see adjustments.

alex-ab, 2017-08-11:
@ehmry: Indeed, it would be good to know where we lost your 4GB, either already by the UEFI memory description, or in grub2 or in the nova kernel or in core.

alex-ab, 2017-08-11:
(or by bender)

m-stein, 2017-08-14:
Looks cool to me :) Added only one minor question in the commit.

alex-ab, 2017-08-15:
@m-stein, @skalk: I adjusted the 26c88e0e036cb8fcfa98b3ae7d80991f64d3c41f according to your reviews, please re-check

alex-ab, 2017-08-17:
After review of @m-stein and @skalk, commits b217178b19b4d76e5a1dc61c26fe2688b9946afa 2f34b5dff7a7700cde6b08d5cb1bcdd3e83c9c50 are ready for base-hw with UEFI

chelmuth, 2017-08-18:
What about a0de778, 03e4f56, and e9aa563?

alex-ab, 2017-08-18:
I would like to test it on real hardware, before going for sel4 uefi support (next week)

alex-ab, 2017-08-23:
@chelmuth: I tested the sel4 commits. You may consider  8b9a76e4e74c82e45da19d06190e15fe02540438 40ac0318c15bacdf2dfbf7630245a98f24e10154 3747190d41dc49f528595b0f0e6e4718ea494050 6981b4a318b06a4fbdb71c24bc31a3398943523e

alex-ab, 2017-08-29:
@chelmuth, @nfeske: please consider 346c9464220578f3dc295c773660a849d9738fc8 for staging

@ehmry: I could reproduce your issue with too few memory, the commit above fixes it for my machine. Please check for your machine.

N8Fear, 2017-08-29:
I'm not sure why, but `prepare_port sel4` fails for me on staging. The error is:
```
sel4  update src/kernel/sel4
sel4  apply /home/n8fear/src/live-sync/genode/repos/base-sel4/patches/root_cnode.patch
sel4  apply /home/n8fear/src/live-sync/genode/repos/base-sel4/patches/config.patch
sel4  apply /home/n8fear/src/live-sync/genode/repos/base-sel4/patches/arm_cache.patch
sel4  apply /home/n8fear/src/live-sync/genode/repos/base-sel4/patches/uefi_mbi2.patch
2 out of 7 hunks FAILED -- saving rejects to file src/kernel/sel4/src/arch/x86/kernel/boot_sys.c.rej
1 out of 2 hunks FAILED -- saving rejects to file src/kernel/sel4/src/plat/pc99/machine/acpi.c.rej
1 out of 1 hunk FAILED -- saving rejects to file src/kernel/sel4/include/plat/pc99/plat/machine/acpi.h.rej
make[2]: *** [/home/n8fear/src/live-sync/genode/tool/ports/mk/install.mk:131: phony/patches//home/n8fear/src/live-sync/genode/repos/base-sel4/patches/uefi_mbi2.patch] Error 1
make[1]: *** [/home/n8fear/src/live-sync/genode/tool/ports/mk/prepare_single_port.mk:42: _install_in_port_dir] Error 2
make: *** [../../../tool/ports/prepare_port:33: sel4] Error 2
```
This is clearly caused by the patches getting applied in wrong order. If I rename the file it works, e.g.:
```
sel4  download https://github.com/seL4/seL4.git
sel4  git Cloning into 'src/kernel/sel4'...
sel4  update src/kernel/sel4
sel4  apply /home/n8fear/src/live-sync/genode/repos/base-sel4/patches/root_cnode.patch
sel4  apply /home/n8fear/src/live-sync/genode/repos/base-sel4/patches/config.patch
sel4  apply /home/n8fear/src/live-sync/genode/repos/base-sel4/patches/arm_cache.patch
sel4  apply /home/n8fear/src/live-sync/genode/repos/base-sel4/patches/noise.patch
sel4  apply /home/n8fear/src/live-sync/genode/repos/base-sel4/patches/benchmark.patch
sel4  apply /home/n8fear/src/live-sync/genode/repos/base-sel4/patches/ioapic.patch
sel4  apply /home/n8fear/src/live-sync/genode/repos/base-sel4/patches/acpi_bootinfo.patch
sel4  apply /home/n8fear/src/live-sync/genode/repos/base-sel4/patches/zuefi_mbi2.patch
sel4  generate sel4.hash
patching file src/kernel/sel4/configs/pc99/ia32/autoconf.h
patching file src/kernel/sel4/configs/pc99/x86_64/autoconf.h
patching file src/kernel/sel4/configs/imx6/wand_quad/autoconf.h
```
It does'nt make much sense to me, why `uefi_...` gets sorted in before `ioapic...` and acpi_boot...`, but that seems to be the case.  Any idea why this happens?

alex-ab, 2017-08-29:
@N8Fear: thanks for reporting! c23a1c260810dfaf3012bf637d96d33794f67ffd should help.

N8Fear, 2017-08-29:
I can confirm that the commit fixes it for me. Thank you!

alex-ab, 2017-08-30:
Initial support is done for the major supported kernels.

closed 2017-08-30 by alex-ab