genodelabs/genode #2243: Remove phys_addr from Dataspace interface

alex-ab, 2017-01-19:
Currently for each Dataspace the physical address can be determined by drivers but also by applications. The physical address should be only obtainable by components which actually has a need for it, like drivers. The platform driver seems to be the right place to mediate about who should see what physical address of DMA memory. In principal the ram_session interface can be extended, like done for phys_start/phys_end by another session connection argument which permits to gather the physical address by some way.

chelmuth, 2019-04-18:
As noted by @tomga on the [mailing list](https://lists.genode.org/pipermail/users/2019-April/006739.html) for RaspberryPi, the information required by device drivers isn't actually the *physical (or CPU logical) address* of the memory but the *bus address* on the bus the device is connected to. Also for this reason we should address this issue and find an appropriate name for the function.

nfeske, 2021-04-07:
I have started to pick up this issue for once. It has almost grown a beard already.

During an offline discussion (including @alex-ab, @skalk, @ssumpf, @cnuke), we identified the following actionable points towards wrapping up this issue.

* The `Cache_attribute` enum has three values CACHED, UNCACHED, WRITE_COMBINED, which is not useful at the API level. It should better be trimmed to the two arguments CACHED and UNCACHED. While I'm at it, I will rename it to simply `Cache`.
* The `Plattform_session::alloc_dma_buffer` should be extended with a `cached` argument.
* We need a new `Region_map::attach_dma` RPC function to avoid the allocation of I/O page tables by the kernel for mere device drivers (when attaching DMA buffers there). I/O page tables should only be created for device PDs. Right now, such allocations are based on looking at the `cached` argument, which is not correct. At a later stage, we may consider introducing a dedicated core session interface for device PDs.
* The `Platform_session::bus_addr_dma_buffer` is too long. I think I'm at fault with picking it in the first place. We settled on `dma_addr` now.

nfeske, 2021-04-07:
Commit https://github.com/genodelabs/genode/commit/07dd42ed05715f25296925650fdf9676bbf2e971 implements the renaming.

Commit https://github.com/genodelabs/genode/commit/3b7703835dbf44a262b1c03c36f91547a9accf6b add `Platform_session::dma_addr` to the x86 platform driver. @alex-ab could you have a look at the latter commit?

nfeske, 2021-04-07:
The imx8 framebuffer driver happens to allocate a RAM dataspace using the write-combined attribute, which caught me by surprise: https://github.com/genodelabs/genode/blob/staging/repos/dde_linux/src/drivers/framebuffer/imx8/lx_emul.cc#L1201 Does that mean we actually need the WRITE_COMBINED case for the Cache attribute enum after all?

nfeske, 2021-04-07:
Commit https://github.com/genodelabs/genode/commit/902d106be260a3bee0f50a3a4ce6f8e7ff6454da does the shortening of the `Cache_attribute` enum to `Cache`.

skalk, 2021-04-07:
> The imx8 framebuffer driver happens to allocate a RAM dataspace using the write-combined attribute, which caught me by surprise: [...]

I'm pretty sure this is a copy'n paste issue. Most of the imx8 framebuffer driver's lx_emul code is a copy of the intel framebuffer driver. @cproc please correct me if I'm wrong. So I'm pretty sure we don't need WRITE_COMBINED. Moreover, the WRITE_COMBINED cache attribute on arm64 at least in base-hw maps to the same page flags like UNCACHED.

cproc, 2021-04-07:
The requirement comes from here:

https://github.com/Freescale/linux-fslc/blob/1d832e20c24583e665fc1731785177bb55b62c46/drivers/gpu/drm/drm_gem_cma_helper.c#L87-L101

alex-ab, 2021-04-08:
> Commit [3b77038](https://github.com/genodelabs/genode/commit/3b7703835dbf44a262b1c03c36f91547a9accf6b) add `Platform_session::dma_addr` to the x86 platform driver. @alex-ab could you have a look at the latter commit?

The commit looks good :+1:

nfeske, 2021-04-08:
> The requirement comes from here:

It still remains rather vague. After consulting with @skalk and @chelmuth, we decided trying to live without the Cache::WRITE_COMBINED enum value and see how we fare.

@cnuke mentioned potential uses of this flag for GPU buffer objects but I'm unable to wrap my head around that. It goes without saying that write-combining for I/O MEM dataspaces will be preserved. So @cnuke's case on x86 could potentially be addressed by creating buffer objects via I/O MEM sessions pointing into the aperture window.

Once we hit an actual wall, we may extend the RAM-allocation interfaces in a deliberate way. At that point we will have developed a good grasp on the requirements and can document the wall we hit. Until then, a tighter interface is better because it leaves less room for interpretation.

nfeske, 2021-04-08:
Commit https://github.com/genodelabs/genode/commit/04d2f18a822aaf6e3736e15e74670f7981369945 imports the `Genode` namespace into the `Platform` namespace, making the platform-driver code easier to read.

nfeske, 2021-04-08:
In https://github.com/genodelabs/genode/commit/b7f56b627128bac4ef6f00fd9e7ab00d0afc9fb2, I added the `Cache` argument to `Platform_session::alloc_dma_buffer`.

nfeske, 2021-04-08:
The next step is replacing core's heuristics about the UNCACHED/CACHED attribute (to decide whether or not DMA page tables need to be populated) on x86 by an explicit `attach_dma_buffer` operation. Intuitively, I considered adding this operation to the `Region_map` as a companion of the existing `attach` and `attach_at` RPC functions first. However, I eventually came to the conclusion that it fits better in the PD session interface, for the following two reasons:

* The `Region_map` is meant as a generic utility. E.g., it is used for the linker area, stack area, and managed dataspaces. In these cases, an `attach_dma_buffer` operation makes no sense.

* Strictly speaking, the use of PD sessions for device PDs is not a perfect fit because the PD session offers a lot of features that do not apply for device PDs. Hence, we will probably end up spinning out a dedicated device-PD session interface that precisely targets device address spaces. When looking towards this direction, I sense that having a `attach_dma_buffer` present directly in the PD session interface will make the future transition simpler.

nfeske, 2021-04-08:
As a third point, the permission check for the managing-system role is straight-forward to accomplish in the context of the PD session, compared to the `Region_map`.

nfeske, 2021-04-12:
Commit https://github.com/genodelabs/genode/commit/4124ecc51300fa4bd0d670ab605bd91cfd21eb1f implements @skalk's suggestion to unify the core-internal `Mapping` type between all base platforms.

@alex-ab on seL4, I had to dumb down the diagnostic message printed whenever a non-memory fault is observed because the seL4-specific fault type can no longer be propagated to the platform PD via the `Mapping` type and the page-fault handler is unaware of PD and thread names. However, we may re-introduce the message later by merging the `Ipc_pager` with the `Pager_entrypoint`.

alex-ab, 2021-04-12:
Ok, i see

nfeske, 2022-02-01:
Commit https://github.com/genodelabs/genode/commit/8d603b2a8042effdc662255b2877945de5dac561 extends the PD session interface with two functions `dma_addr` and `attach_dma` as another step. The ambiguous use of the UNCACHED attribute of RAM dataspace is now. The `dma_addr` functions in not used yet.

The next step would be adding the use of `Pd_session::dma_addr` to the x86 platform driver and adjusting the individual drivers to use this interface instead of calling `Dataspace::phys_addr`.

nfeske, 2022-02-02:
Fixup https://github.com/genodelabs/genode/commit/5de5c18d32e42878b7215b4ea27a555540836fde changes the platform driver(s) to use `Pd_session::dma_addr` instead of `Dataspace::phys_addr`.

nfeske, 2022-02-02:
Commit https://github.com/genodelabs/genode/commit/b7fd199cd5ea31b48e0d5f1a80261737c6b83b3c updates quite a few drivers.

There are still some drivers I'm hesitant to touch and would prefer leaving the adaptation to developers familiar with the code.

In particular, @cnuke or @ssumpf could you take care of updating the Intel GPU driver?

The virtio drivers are also raising some questions. Right now, the communication buffers allocated in virtio/queue.h are plain RAM dataspaces, allocated from the PD session, not the platform driver. Hence, those buffers are strictly speaking not DMA-capable. So the buffers should better be obtained from a platform connection that also provides their DMA-address information. What makes the situation a bit hairy right now is that the drivers support both PC and ARM, which currently use different platform-session APIs. @skalk could you investigate a possible update strategy for this driver?

Finally, there is the old-school block/component.h API, which is currently still used by several SD-card drivers (that haven't been updated to the `Block::Connection::Job` API yet).

nfeske, 2022-02-03:
It turns out that the following DDE-linux-based ARM drivers still allocate DMA buffers as plain RAM dataspaces, not using the platform session. The platform session is merely used within lx_emul.cc.

* fec_nic_drv
* imx6q_sabrelite_usb_host_drv
* panda_usb_host_drv
* arndale_usb_host_drv

I think all of them copied the same pattern. @skalk could you take care of adjusting those?

jschlatow, 2022-02-03:
For the record, the zynq_nic_drv requires an update as well. I'll take care of that.

cnuke, 2022-02-03:
@nfeske please merge commits [6eb922](https://github.com/cnuke/genode/commit/6eb922c7248ecddd1d8a7ae8e9e93f0260cecd94) and [be05d85](https://github.com/cnuke/genode/commit/be05d858f7e7f716eae08ce5e64c53074a4aa74f).

skalk, 2022-02-03:
@nfeske commit 029d9fae599b26490962d6ea84fb14c601f69d4d solves the failures of the usb_host and fec drivers, commit ac9255d934414bbb52951daab28e2baf2c5539c2 reworks the virtio drivers to use the platform driver instead of the Dataspace::phys_addr calls.

nfeske, 2022-02-03:
Thanks a lot @cnuke for the patches for the GPU and audio drivers!

nfeske, 2022-02-03:
Thanks @skalk for the welcome fix of the fec_nic_drv and the quick adaptation of the virtio driver. I'm very grateful for your support!

nfeske, 2022-02-03:
@skalk I noticed one remaining call of `Dataspace::phys_addr` by the virtio driver at https://github.com/genodelabs/genode/blob/staging/repos/os/include/virtio/queue.h#L254.

skalk, 2022-02-04:
@nfeske sorry, I was too much into it when removing the first two places to recognize the third place of phys_addr. The fixup commit e53a7538e466df0831ec33d862aaa98285a12709 eliminates the remaining call.

nfeske, 2022-02-04:
Thanks a lot Stefan!

jschlatow, 2022-02-08:
@nfeske, I changed the zynq_nic_drv in the genode-zynq repo to not use phys_addr() anymore.

commit 84f5d486 (2022-02-11): Remove Dataspace::phys_addr RPC function

The official way to obtain DMA addresses for RAM dataspaces is
the RPC function 'Pd_session::dma_addr' now. User-level device drivers
should not call this function directly but use the 'Platform_session'
interface of the platform driver instead.

Fixes #2243
closed 2022-02-15 by nfeske