genodelabs/genode #2249: Child: invalid binary name gives no error.

BorisMulder-CSL, 2017-01-20:
Whenever a new Genode::Child is created, and the binary name provided in the child policy is invalid (no binary with that name exists), the constructor of Child will simply return without having initialized its env-dependent members.
In base/src/lib/base/child.cc, the constructor will call "_try_construct_env_dependent_members", and inside, will check if the capabilities for the environment sessions are valid. Then, if one of them isn't, it will return (without giving any error message), since _binary.cap().valid() will fail. This causes some objects, such as the thread and heap not to be constructed. 

So if the heap() function is called, for instance right after the child object is constructed, it will throw a Deref_unconstructed_object, causing an abort. In any case, the child does nothing and gives no error.

Perhaps having a typo in a binary name should give an error message stating that the binary name is invalid right when the binary name is verified? Now it took me a lot of time to find out that it was simply misspelled.

nfeske, 2017-01-20:
Thanks for reporting the detailed analysis.

This problem has been discussed in #2214 and is addressed by the commit https://github.com/genodelabs/genode/commit/73837ef56a075519774404bca7132879f92993e6  on the master branch.

nfeske, 2017-02-28:
Fixed in Genode 17.02.

closed 2017-02-28 by nfeske