genodelabs/genode #2250: init: remove outdated `<if-arg>` mechanism

nfeske, 2017-01-20:
The `<if-arg>` mechanism was an ad-hoc solution to route session conditionally, dependent of arbitrary session arguments. If was later superseded by our labeling mechanism that we nowadays use consistently. We should relieve init from the legacy `<if-arg>` baggage and adjust the few run scripts that still depend of it.

trimpim, 2017-01-21:
We rely heavily on the <if-arg ..> mechanism. We have different components that provide the same service, but with different sub-services.
Can the new mechanism also be used, if the sub-services aren't differentiated by the key label but by something else (eg. "subservice")? Otherwise I have to plan accordingly the changes on our code.

nfeske, 2017-01-21:
I am sorry to hear that this change makes your life difficult. In hindsight I regret the weak moment when I added the `<if-arg>` mechanism to init. It was an ad-hoc solution to a problem that I did not fully thought through at the time. I deliberately left the feature undocumented but as often with such hacks, their use eventually proliferates anyway.

Regarding your question, no, the session routing can depend only on the service name and the session label. Other session arguments are solely meant as session-construction arguments supplied to the server, or denote the session quota. Please note that the decision to boil down the criterion for the session routing to the service name and session label is not an arbitrary decision from my side. It is an important prerequisite for future tools and user interfaces for creating and managing Genode systems.

How to proceed best? Admittedly, I am unfamiliar with the notion of a "sub service". Would it be possible to model your use case via session labels and/or service names? if not, I'd love to understand the specific problem that you cannot solve this way. So I become able to consider it in the new version of init. That said, given the system scenarios that I encountered so far, I am quite hopeful that the labeling mechanism may actually suffice when looking at the problem from a slightly different angle. E.g., in some situations, it is sensible to introduce a convention for session labels such as prefixing labels with a certain property (like a tag). So when creating an init configuration with explicit session routes, one immediately distinguishes the different types of labels (according to their encoded properties).

trimpim, 2017-01-23:
The whole thing comes from a software that we ported to Genode.
After sleeping over the issue, I think the changes will be minor, just rename "subservice" to "label" and change the routes to use the new easier to understand (and less typing) syntax.

nfeske, 2017-01-23:
@trimpim Thanks for the confirmation that I can go on without putting you in jeopardy. I am relieved. ;-)

nfeske, 2017-02-28:
Removed in Genode 17.02

closed 2017-02-28 by nfeske