genodelabs/genode #2260: Trace test must be revised

chelmuth, 2017-01-27:
The test in *repos/os/run/trace.run* has several issues.

1. The test requires *rpc_name*, which can't be built for riscv as the linker chokes with a segmentation fault. The planned tool-chain update should resolve this.
2. The core `Trace::Session_component` does not use the `Session_policy` and, therefore, does not support `label_prefix`. This can easily be fixed.
3. The core implementation of the trace-policy handling uses a slab cache for policy objects but the types for slab dimensioning and the actual object instance seem to mismatch.

That's what I found out this morning, but as I did not fix 3) there may be more hidden issues.

cnuke, 2017-01-30:
5ffc7f1 makes the test work again, although I'm puzzled why the explicit namespace has to be used for the compiler to calculate the right size.

cnuke, 2017-01-30:
I reworked the test and now the run script uses the `null` policy module (so we can enable the test in the autopilot).

ValiValpas, 2017-01-30:
for the record: this is related to #1508

cnuke, 2017-01-30:
@ValiValpas thanks for the reminder, I completely missed this old issue :+1:

chelmuth, 2017-01-31:
If I understand it correctly dc6247d qualifies as fix for #1508, right?

cnuke, 2017-02-01:
4fd130e should fix it but I did not verify it.

nfeske, 2018-03-01:
Fixed in master.

closed 2018-03-01 by nfeske