genodelabs/genode #2263: Cleanup of the VFS block file system

nfeske, 2017-01-28:
I think that the VFS block file system is not in a good shape. I would like to see it improved in several respects:

* The `_block_io` method should be replaced by `_read` and `_write` methods. Handling both operations in one method makes things needlessly complicated and inconsistent. I.e., the misleading error message "Could not read block(s)" that could also appear on write operations, or the use boolean `write` argument, and the optional `bulk` argument.
* The handling of partial block accesses is hard to comprehend and depends on questionable conditions. In particular, I was wondering about `(count % _block_size) >= 0`. Isn't this always true?
* It should use the block session asynchronously, once the VFS supports this.
* It should use the `blit` library instead of byte-wise memcpy.

cnuke, 2025-06-13:
* Commit [8c09ecb](https://github.com/cnuke/genode/commit/8c09ecb03c358c6e87dc5208af67a5a9b39f7692) changes the *block_count* option to a more meaningful *io_buffer* option that better reflects the intend.
* Commit [0cb6575](https://github.com/cnuke/genode/commit/0cb657574850e59c6f1d2cdc9dbcaaaca79e71c1) allows for using the static `Block` session information in a cached way (mainly used for getting rid of the various `_info` members that often are in place because an RPC would be to costly).
* Commit [5ca3226](https://github.com/cnuke/genode/commit/5ca3226c4bd53026944da373fded22231e7ffde5) renovates the plugin in a plain way (no block-caching and a fairly minimal amount of logic to deal with unaligned and/or partial requests) and uses the Job API for the heaving lifting.

Judging by my testing the new plugin performs similar to the old one for the non-concurrent case where it delivers the same throughput while incurring a similar system load. If it is used concurrently, e.g. using a `fio` test that is spawning multiple jobs (openeing the file in multiple pthreads and issuing the *same* request multiple times from them in the same order) the throughput is much higher as the configured I/O buffer is thoroughly utilized. This property could come in handy when using multiple VFS handles to map requests for batching in an *aio(4)* context.

nfeske, 2025-06-19:
Thank you very much for this thorough rework! I merged the three commits to staging.

The change of the default I/O buffer size is very welcome. Do you happen to have tested Sculpt's block-device operations with the changed default (format device/partition, extending a partition) to see if the quota of the spawned utilities still suffices?

nfeske, 2025-06-19:
Excellent! Your fixups went to staging right away.

nfeske, 2025-06-21:
I  remember struggling with similar issues (32 vs. 64 bit) while reworking the block_tester. After several iterations, my final solution worked out quite well. @cnuke how do you think about using the types as defined there? See https://github.com/genodelabs/genode/blob/master/repos/os/src/app/block_tester/types.h#L25. Should the types be equally well applicable for the VFS-block plugin, we may consider moving them to a central place.

closed 2025-08-28 by chelmuth