genodelabs/genode #2265: Genode::String x(Genode::Hex(value)) produces not expected output

alex-ab, 2017-01-30:


alex-ab, 2017-01-30:
The commit above triggers the issue:

```
[init -> test-log] hex range:          [0e00,1680)
[init -> test-log] empty hex range:    [0abc0000,0abc0000) (empty!)
[init -> test-log] hex range to limit: [f8,ff]
[init -> test-log] invalid hex range:  [f8,08) (overflow!)
[init -> test-log] negative hex char:  0xfe
[init -> test-log] positive hex char:  0x02
[init -> test-log] multiarg string:    "parent -> child.7"
[init -> test-log] snprintf replacement: 0x
[init -> test-log] Test done.

Line 7 of output is unexpected
 expected: '[init -> test-log] snprintf replacement: 0x3'
 got:      '[init -> test-log] snprintf replacement: 0x'
```

alex-ab, 2017-01-30:
@nfeske: have two different ways to solve the issue, however I'm unsure which of the commits are the right one. I would guess variant 1 and variant 2 just workaround the issue ?

nfeske, 2017-01-31:
@alex-ab thanks for uncovering and fixing this issue. I agree that the variant 1 is the better fix. I'd like to add the test case to the log test. However, as a minor detail, would you fix the horizontal alignment by adding one space to the previous lines or by shortening the new line accordingly?

alex-ab, 2017-01-31:
@nfeske: Ok, done.

commit dd5592cc (2017-01-31): Fix String(Hex()) usage

- additional add test casye to log.run

Fixes #2265
closed 2017-02-07 by nfeske