genodelabs/genode #2268: fb_bench: make it run on all platforms

konglomerat-org, 2017-01-31:


m-stein, 2017-01-31:
@chelmuth The commits
e53f67a fb_bench: start QEMU with -nographic
5b8e63d fb_bench: mark wand_quad as not supported
c1511e6 fb_bench: correct "platform not supported" message
0ae692d fb_bench: add gpio driver when supported
should fix all platforms expect sel4 and pistachio. For pistachio it might be best to make the test signal driven instead of doing it completely in the initial thread. Pistachio finishes successful but it needs much longer as other platforms and the measured value are somehow bogus. On sel4 we get massive warnings...
_lookup: page-table lookup failed
...and some warnings...
flush page table entries - mapping cache full - PD: init -> test-fb_bench
... and then an error...
Uncaught exception of type 'Genode::Bit_allocator<16384u>::Out_of_indices
... but the first test finishes. Maybe it can be fixed (later).

chelmuth, 2017-01-31:
Thanks, merged all four commits.

m-stein, 2017-02-01:
I've added further commits according to the test results of the last night:
8f57578 gpio_drv: get rid of specs foc_rpi and hw_rpi
b069872 fb_bench: do not test on zynq
6751c61 create_builddir: do more elaborate sanity checks

They should fix problems with hw+rpi, foc+rpi, and zynq.

m-stein, 2017-02-08:
The following patches should solve recent problems with imx53_qsb and panda
931cacf fb_bench: do not test on i.MX53 + Fiasco.OC
09c0d45 fixup "gpio_drv: get rid of specs foc_rpi and hw_rpi"

nfeske, 2017-02-09:
I merged the first commit. Before merging the second one, however, let me suggest a stylistic change. The current version of the commit uses a Tcl variable like this:

```
set gpio_drv hw_gpio_drv
...
append_if [have_spec gpio] config "
	<start name=\"$gpio_drv\">
    ...
```

I would rather prefer to promote the (functional-programming style) pattern of using a `proc` instead:

```
proc gpio_drv { } { return hw_gpio_drv }
...
append_if [have_spec gpio] config "
	<start name=\"[gpio_drv]\">
    ...
```

Please note how this pattern is already consistently used in `base/run/platform_drv.inc`.

m-stein, 2017-02-09:
@nfeske I've updated the fixup to c7e5b5d7c8969336429c3751c13fa86e9085d306

m-stein, 2017-02-10:
@nfeske The following commits are a first approach to the sel4 problem:
67b5214 Do not merge "base-sel4 core: try to flush page entry selectors as well"
7e533cf Do not merge "base-sel4 core: flush page cache from outside of 'map'"
7177108 Do not merge "base-sel4 core: use Vm_space::Type instead of bool"

The last one is the only one that contains "real"  behavioral changes.

alex-ab, 2017-09-05:
The test results on 2017-09-03 where all good for fb_bench, may we close it ?

m-stein, 2017-09-06:
In an offline discussion with @alex-ab, we found that the "Do not merge" commits for SeL4 were fixed by commits of other issues in the meantime. Thus, this issue can be closed.

closed 2017-09-06 by m-stein