genodelabs/genode #2284: signal_test: `Wait_and_dispatch_test` does not return

ssumpf, 2017-02-21:


ssumpf, 2017-02-21:
@nfeske: please move 8eb29ed and 8f0cccd to staging.

m-stein, 2017-02-24:
@nfeske Please merge when things have settled:
98dcbbe hw: fix race in signal dispatching

nfeske, 2017-02-24:
The patch looks too good to be not merged right away. ;-)

ssumpf, 2017-02-25:
@nfeske: Please merge 21e6c66 to staging

m-stein, 2017-02-27:
@nfeske Added another commit:
8d71b47 signal test: add stressful nested test

nfeske, 2017-02-28:
Fixed in master.

closed 2017-02-28 by nfeske

cnuke, 2017-03-01:
@chelmuth or @nfeske please merge 0b4443f to staging.

cnuke, 2017-03-01:
@chelmuth or @nfeske please merge d5dc9c1 to staging (second time is the charme).

alex-ab, 2017-03-08:
@nfeske, @chelmuth: please consider bd8a903f41312a560dc555a4dc79cdf6b335b210 for staging, it fixes issues during destruction to the 9th test

alex-ab, 2017-03-08:
@chelmuth: please consider 7810decfbf238d082b4897c53aa94d19d8a5bbb0 for staging instead of the previous commit

chelmuth, 2017-03-08:
Nice, merged to staging just now.

commit cb6cf251 (2017-02-26): fixup: 'hw: fix race in signal dispatching'

Fixes race between signal lookup (EP) and signal thread, where the
signal thread might see state ENTRYPOINT, the EP finds a pending
signal, calls _dispatch, while the signal thread calls
cancel_next_await_signal leading to very dangerous semantics.

definitely fixes #2284