genodelabs/genode #2298: nova: use 'Native_cpu' component for thread initialization

cproc, 2017-02-27:
Pass the thread type and exception base to core with a 'Native_cpu' component instead of enhancing the 'Thread_state' class.

This solves problems when using binaries with different kernels and the wrong variant of the 'thread_state.h' file was included (happened with the 'cpu_sampler' component).

cproc, 2017-02-27:
The commit seems to work with cpu_sampler and gdb_monitor, but I can't guarantee that it doesn't break something else. The gdb_monitor changes depend on the cleanup commit on the same branch (https://github.com/cproc/genode/commits/issue2298).

alex-ab, 2017-02-27:
I would encourage not to add another 300 lines of code and instead remove 30 lines. Why ? It is easier to read, it is easier to grasp, no virtualization headaches (cpu_sampler/gdb), less code in the TCB. And it breaks nothing, I guarantee.

nfeske, 2017-02-28:
@alex-ab the direction to introduce a NOVA-specific native-cpu interface came from me and I'd like to explain the rationale behind it:

The interface functions that we expose via the generic Genode API should have an unambiguous meaning across all base platforms unless it is infeasible to achieve (like remote address space manipulation on Linux). NOVA requires a very special thread startup procedure that deviates from the generic interface. Now there are three possible options how to deal with this situation:

* Twisting the generic interface in creative ways such as carrying additional information in supposedly unused bits of arguments that are already there. In essence, this is synonymous to passing arguments via a covert channel. In practice, those deviations from the documented interface may be immediately convenient (because the official interface is left unchanged) but in the long term, they bite us. Please let us remember the NOVA-specific addition of a receive window to the capability type, which was a solution of the same pattern, and caused me serious headache much later. In fact, the addition of the exc_sel, vcpu, and global members in the NOVA-specific thread state already was misguided by the same wrong pattern. The thread-state was effectively misused to carry additional thread-start arguments.

* Making the peculiarities of platforms explicit where they deviate from the norm. This is precisely the purpose of the native-cpu interface. It is the designated place for platform-specific additions and there exists an established pattern how to use it (see base-linux and base-foc). When talking about being explicit, I go as far as introducing dedicated types for different method arguments to make the argument use absolutely unambiguous and thereby not prone to accidental mistakes (like passing arguments in the wrong order, or clearing the wrong bit of a dumb integer argument). This is why I recommended @cproc to introduce `Exception_base` an `Thread_type` as types. Given the new interface, it becomes very clear where NOVA deviates from the generic API and where to look for these peculiarities, and where to look for the documentation of them(!). In contrast, your proposed patch documents the misuse of sp and ip merely in the implementation of `platform_thread.cc`. I fail to see how this would be easier to read as you claim. Regarding the TCB complexity, I'd rather prefer elaborate (in terms of lines of code) strongly typed an well documented code that uses established patterns than a few lines of bit fiddling or pointer arithmetics or similar C-ism. Experience tells us very clearly, in which kind of code the bugs usually hide.

* The most elegant solution would be to remove the NOVA-specific thread-startup procedure from the API altogether. Core could actually allocate the exception window for the new thread in a core-managed portion of the PD's cap-selector space. The VCPU case could to removed by implementing core's `Vm_session` service for NOVA. This leaves solely the distinction between local/global threads. But I am positive that we might even find a good solution for that one as well (e.g., considering that a `Genode::Entrypoint` is always a pair of a local and global EC, we may create both when `Cpu_session::create_thread` is called).

To me, it is clear that the third way is the best. But it is also the most elaborate one. So it is infeasible to go for @cproc to fix his immediate problem. This leaves only the first two options as viable immediate approaches. I strongly prefer the second one as it adheres the principle of least surprise, does not twist the API in unintended ways, and follows an established pattern instead of being creative.

alex-ab, 2017-02-28:
I know your arguments, and why things are as they are. The point is, that if you have to add 300 lines of code in order to transfer 1-2 words for thread creation, there is something wrong. I argue that the generic thread interface is not generic enough to pass some simple thread creation arguments (like type of the thread). 

Since the third option is the one to actually go for, I would not use a over-complicated second approach and adding maintenance burden.

nfeske, 2017-02-28:
@alex-ab You overestimate the impact of cproc's patch on the code complexity of base-nova. I just gave sloccount a try:

* base-nova at staging: 5,710
* base-nova with cproc's patch: 5,784

Hence, the difference is actually 74 lines of added lines of code, where some of the lines are on account of the stronger typing (which is arguably more verbose yet more robust).

The choice between option 1 or 2 comes down to a matter of _direction_. Do we want to go in the direction of being always explicit about semantics, attaining self-documenting code, leveraging strong typing? I hope you agree that this is the direction we should strive for to make Genode rock-solid. Even if we will eventually abandon option 1/2 for option 3, we should still not walk backward, not even for a stop-gap solution.

alex-ab, 2017-02-28:
You don't count the burden on the server using the interface, with it it is 300 lines addition.

>  I hope you agree that this is the direction we should strive for to make Genode rock-solid.
That is direction 3 and a generic interface that is actual generic.

nfeske, 2017-02-28:
Actually, I do not understand why the CPU sampler intercepts the CPU-thread interface. Without this interception, the native-CPU wrapper would not be needed.

That said, the code complexities of neither GDB monitor nor the CPU sampler are critical. Both are debugging tools that don't taint the TCB after all.

> That is direction 3 and a generic interface that is actual generic.

We agree here. Please don't take me wrong. I have not pretended that the NOVA-native-CPU interface was generic. It clearly isn't. I have said that it is a clean solution for a problem that cannot be solved in a generic way right now.

commit f5a54c04 (2017-03-01): nova: use 'Native_cpu' component for thread initialization

Pass the thread type and exception base to core with a 'Native_cpu'
component instead of enhancing the 'Thread_state' class.

Fixes #2298
commit ca2871e2 (2017-03-15): nova: use 'Native_cpu' component for thread initialization

Pass the thread type and exception base to core with a 'Native_cpu'
component instead of enhancing the 'Thread_state' class.

Fixes #2298
closed 2017-03-15 by chelmuth