genodelabs/genode #2299: livelocks in cxa guard when using threads with various priorities

alex-ab, 2017-03-01:


alex-ab, 2017-03-01:
I extended the showcase test to also trigger multiple cxa guards concurrently in use by several threads.

alex-ab, 2017-03-01:
The attached fix uses a Genode Semaphore to handle the contention cases, where the blocked threads gets waken up/resumed in a stacked way one by one.
I will try to come up with another variant, as discussed with @chelmuth, which will instead resume all at once. Then you may vote for your favorite ;-)

alex-ab, 2017-03-02:
As promised, a second version. Both commits solve the issue for me. Choose wisely - 53048b755625dc3cf40dc98871744883f3098ad8 vs 91dfdd93f6f96c2b7ee482dd600fd1d3d4b08cdb.

alex-ab, 2017-03-02:
Sorry, wrong commit referenced.

As promised, a second version. Both commits solve the issue for me. Choose wisely - 6d323690030619a8a66983a51f1492b576c4e06b vs 91dfdd93f6f96c2b7ee482dd600fd1d3d4b08cdb.

alex-ab, 2017-03-03:
I added another version which uses the newer Registry class instead of manually List/Lock combination.

chelmuth, 2017-03-06:
I merged variant B into staging because of the open issues with C.

alex-ab, 2017-03-14:
@chelmuth: please consider the last 4 commits of https://github.com/alex-ab/genode/tree/issue_2299_c_staging for staging

commit 508a8ec1 (2017-03-06): fixup "base: extend cxa guards to support threads on various priorities"

Fixes #2299
commit deadfbcd (2017-03-06): base: extend cxa guards to support threads on various priorities

Fixes #2299
commit 92f8517f (2017-03-06): base: extend cxa guards to support threads on various priorities

Fixes #2299
commit d896022d (2017-03-06): fixup "base: extend cxa guards to support threads on various priorities"

Fixes #2299
closed 2017-03-15 by chelmuth