genodelabs/genode #2300: Genode::Lock - wrong threads getting into critical sections and threads getting lost (never waked up) once in contented lock

alex-ab, 2017-03-01:


alex-ab, 2017-03-01:
The testcase shows the issue

```
...
[init -> test-thread]  thread 'lock_low1' started
[init -> test-thread]  thread 'lock_low2' started
[init -> test-thread]  thread 'lock_low3' started
[init -> test-thread]  thread 'lock_low4' started
[init -> test-thread]  thread 'ep' - I'm the lock holder - take lock again
[init -> test-thread]  thread 'lock_low5' started
[init -> test-thread]  thread 'lock_low5' got lock but somebody else is within critical section !?
[init -> test-thread] Error: Uncaught exception of type 'int'
[init -> test-thread] Warning: abort called - thread: lock_low5
```

chelmuth, 2017-03-01:
For clarification: This issue happens in the case that the lock owner calls Lock::lock() again.

commit 77f92971 (2017-03-01): base: handle twice Lock::lock calls correctly

to avoid loosing threads in the applicants list if lock owner calls 'twice'
Lock::lock().

Fixes #2300
commit 25ac3cdd (2017-03-15): base: handle twice Lock::lock calls correctly

to avoid loosing threads in the applicants list if lock owner calls 'twice'
Lock::lock().

Fixes #2300
closed 2017-03-15 by chelmuth