genodelabs/genode #2303: VFS pipe plugin

ehmry, 2017-03-01:
Unix pipes for are currently implemented as special plugin to libc, which limits the termination of both endpoints to a single address space. By moving pipes to a VFS plugin, communication channels may be shared between components using the VFS server, and with or without libc at each end.

ehmry, 2017-03-07:
Progress on the pipe is blocked until non-blocking writes are supported. The problem is that reads can come up short, but writes must block until a reader consumes data from the pipe buffer.

chelmuth, 2017-03-08:
I had a close look at _man 7 pipe_ to understand your read/write requirement stated above. First, I've the feeling the pipe plugin is more like a FIFO because the _pipe_ gets a name in the VFS and can be opened by other components when the plugin in instantiated in the VFS server. Regarding read/write I'd like to quote the man page.

> If a process attempts to read from an empty pipe, then read(2) will block until data is available. If a process attempts to write to a full pipe (see below), then write(2) blocks until sufficient data has been read from the pipe to allow the write to complete.

IMO this only means that a write may not finish if the pipe buffer has not enough available space for the write operation. In the case of the fs-file-system VFS plugin, we may treat the packet-stream buffer as the pipe buffer, writes will block if the packet stream becomes full.

chelmuth, 2017-03-15:
After skimming the commit above I got the feeling it's not ready for merge, right? For example, I was wondering about the purpose of the `lock` in the `Io_response_handler`.

ehmry, 2017-03-15:
No, not ready yet, I need to finish compatibilty with the FS session.

The lock in Io_response_handler is there so that the blocking write call can pause and allow other threads into the VFS, without it the two threads of the libc_pipe test will enter at the same time and the ring buffer gets messed up.

ehmry, 2017-03-15:
I'm not sure how to procede with this, the lock cause a deadlock with FS session, yet the VFS still needs some protection, and I don't fully understand the role of ``Io_response_handler``.

If the pipe is not a priority I would look into other VFS work to gain more perspective.

chelmuth, 2017-03-15:
It sounds like a thorough look into the VFS and the libc plugin is worthwhile. Regarding `Io_response_handler` you may think of the `tick` function we had during our early lxip developments. The response handler gives upper layers a chance to react on I/O activity.

chelmuth, 2017-09-29:
Rediscovered this issue just now and vote for updating the code for integration.

ehmry, 2017-09-29:
Yes, async writes and sync do make this more practical.

NobodyIII, 2018-02-12:
What's the status of this? A VFS pipe plugin would be very useful. That would remove the need for the messy hack I was planning on using. A single plugin would be much more elegant than emulating a pipe using file_terminal, a terminal server, and terminal_fs.

NobodyIII, 2018-11-22:
I'm working on my own implementation using a FIFO VFS plugin, with a simple corresponding libc plugin. When it's ready, should I put it here or create a new issue?

chelmuth, 2018-11-29:
@NobodyIII better post and discuss here to prevent duplicates

ehmry, 2019-06-04:
I have a new implementation of the VFS pipe plugin at f7cfa8cefbfe319924be21327470c62e6824d78b, which depends on a write completion loop patch at efe39ab0775bca6bac778fbda43e0c011079aaa8. The `test-libc_pipe` package has been updated to test both local pipes and pipes across a File_system session.

The implementation is less than ideal, but is an intermediate step until a more thread-friendly libc. Whereas the old plugin deblocked threads directly using a semaphore, this plugin use an I/O signal to resume blocked threads. This is to stay consistent with the rest of the VFS, which assumes only a single thread may enter the VFS.

The current libc `io_progress_response` handler installed on all VFS handles simply resumes all blocked threads when called, and is only called during I/O signal dispatching. An alternative solution would be to directly block and unblock pthreads when the `io_progress_response` handler is called across threads, and then the pipe plugin would not need to send a signal internally. The public interface of the VFS is ready for such changes within the libc.

NobodyIII, 2019-06-04:
@ehmry Thanks for figuring this out. I'll give it a try, and then see whether I can adapt my own FIFO plugin to work with your improvements. I was thinking the File_system interface had to be changed to allow partial reads and writes, but you realized it could be done at the VFS level. Nice work!

NobodyIII, 2019-06-04:
@ehmry I wasn't able to cherry-pick your plugin commit. However, I was able to cherry-pick your write-loop commit, and then adapt my plugin to work with that. However, you don't have a read loop, which my plugin needs. I'll write a commit for that.

chelmuth, 2019-06-05:
> I have a new implementation of the VFS pipe plugin...

This looks promising, great. I'd much appreciate a README describing the workings of the pipe_fs incl. semantics of */new*.

> The implementation is less than ideal, but is an intermediate step until a more thread-friendly libc.

So, the libc is thread-unfriendly...

> Whereas the old plugin deblocked threads directly using a semaphore, this plugin use an I/O signal to resume blocked threads. This is to stay consistent with the rest of the VFS, which assumes only a single thread may enter the VFS.

I'm curious: Do you envision to change that?

> The current libc `io_progress_response` handler installed on all VFS handles simply resumes all blocked threads when called, and is only called during I/O signal dispatching. An alternative solution [...]

The current implementation reflects the following property: Only the thread itself (by variables on its stack) knows if an I/O progress may deblock it. So, all threads must be woken up to check their blocking states. Your text hints that you want to explicitly record the blocking state such that a deblocker may directly unblock a thread. Or do I get you wrong?

ehmry, 2019-06-05:
>> The implementation is less than ideal, but is an intermediate step until a more thread-friendly libc.

> So, the libc is thread-unfriendly...

By this I mean that the interactions between the initial application thread, additional pthreads, and the `io_progress_handler` may be problematic. For example, if the main thread were to stay busy in a computational loop while a pthread tries to perform I/O, there can be problems, which is a different issue in itself.  

> > Whereas the old plugin deblocked threads directly using a semaphore, this plugin use an I/O signal to resume blocked threads. This is to stay consistent with the rest of the VFS, which assumes only a single thread may enter the VFS.
> 
> I'm curious: Do you envision to change that?

I don't want to change the VFS to support multiple threads, but to that end the VFS should be fully asynchronous, which it is. I used a signal to be sure that unblocking behaves as much as possible like the existing VFS (just like a pipe over a File_system session).

> The current implementation reflects the following property: Only the thread itself (by variables on its stack) knows if an I/O progress may deblock it. So, all threads must be woken up to check [...]

I think that in the case of pthreads we may be able to put a semaphore into the `Libc::Suspend_functor` that is unblocked directly when the VFS handle owned by the blocked pthread has its  `io_progress_response` method called from another thread. The active `Libc::Suspend_functor` would probably still need to be registered globally to be unblocked by signals.

ehmry, 2019-06-05:
I bring up unblocking threads using semphores and avoiding suspend cycles because then the main thread can unblock a pthread without a suspend. This is why I had to modify the libc_pipe test to suspend the main thread to allow the pthread to complete its read from the pipe (of course if `join` were implemented, then it would suspend the main thread).

```
	/* pthread_join() is not implemented at this time */
	while (!reader_finished) {
		/* XXX: sleep to suspend main thread and dispatch libc events */
		usleep(500000);
	}
```

ehmry, 2019-06-05:
I have a README fixup at 94af2fc5d1d8a14cc25aaa25948c46acb6e0ef54 and two patches for write loops at lwIP and Lxip plugins at 12c8f8177eb6430196f40e0b200da77618b6b344 and 4d1022608768b5e1ab4b82a9f1e660f862ac3c6b (#2971).

chelmuth, 2019-06-05:
Thanks for the README fixup, expectedly it reads much like the socket-fs documentation.

Regarding the direct deblocking I still not get how you want to address blocks on poll/select with multiple file descriptors, but this we may reason out in an offline discussion. But nevertheless an exciting direction, which should definitely include a solution to pthread locking and join.

NobodyIII, 2019-06-05:
My FIFO/pipe plugin creates a read handle and a write handle, which are
handed out in the open() function. Calling close() on one of the handles
releases it, allowing it to be opened again later. This simplifies the
unblocking mechanism.

Once I resolve a strange segfault, and get my FIFO to work when accessed
via the File_system interface to the VFS server, I'll make my code
available.

ehmry, 2019-06-17:
It turns out that `pthread_join` is implented, the libc_pipe test had a stale comment, fixup at d63e7e2e297776aa5fe450535cd268f49eb4800b.

nfeske, 2019-06-17:
Merged the fixup as 374c51f77a53730a82fcc214ed63a0085cea70ff. Note that the patch did not cleanly applied to staging. I adjusted it a bit.

chelmuth, 2019-07-09:
Please correct me but it seems the original pipe plugin did not made it to staging yet. So, I rephrased the commit message of 374c51f77a53730a82fcc214ed63a0085cea70ff to *Use pthread_join in test/libc_pipe*. @ehmry please rebase the feature as one commit after the Sculpt release.

ehmry, 2019-07-09:
Rebased at cff389758eb2c415cb3aefa2cfb5b52e35916947 and 8509e35e621343d3a99bed6911f39539ac3cdbb8.

chelmuth, 2019-07-11:
Last night depot_autopilot libc_pipe failed on all platforms. Does this stem from the VFS pipe plugin itself or my revert of 0f9fa5bf8499267bb99a9932c0b69cd18713fe8e?

NobodyIII, 2019-07-11:
Did it fail with a segfault? Because that's what happened with my FIFO
plugin.

ehmry, 2019-07-12:
Without looking at any logs I would assume that if 0f9fa5bf8499267bb99a9932c0b69cd18713fe8e were reverted the test would stall.

chelmuth, 2019-07-12:
Thanks for the investigation. I applied the patch to use *vfs_rump* and removed the revert. The weekend will reveal any potential regressions.

nfeske, 2019-11-03:
Commit https://github.com/nfeske/genode/commit/bc84f10bd4aa0ebafcee52e50bdb01960564ec47 is rebased on staging (adapted to vfs and libc changes) and extended with EOF handling.

Combined with https://github.com/nfeske/genode/commit/0ef385ea5b8aa71e6c0cc7825ba0e36b115aed56 and https://github.com/nfeske/genode/commit/b52a5eb0e938d028b7d29ba4038cd5c6a7c7009f, the vfs_pipe plugin allows the use of pipes in the bash.run script.

commit dc9c45d0 (2019-07-09): New VFS plugin for emulating POSIX pipes

Add a new plugin for creating pipes between pairs of VFS handles. It is
intended to replace the libc_pipe plugin, one of the last remaining libc
plugins.

In contrast to the libc_pipe plugin, this plugin defers cross-handle
notification until I/O signal handling rather than block and unblock
readers using a semaphore. This is a performance regression in the case
of multiple threads blocking on a pipe, but shall be an intermediate
mechanism pending renovations within the libc VFS and threading layers.
As a side effect, threads blocked on a pipe might no [... cut]
commit 0ea825b8 (2019-07-12): New VFS plugin for emulating POSIX pipes

Add a new plugin for creating pipes between pairs of VFS handles. It is
intended to replace the libc_pipe plugin, one of the last remaining libc
plugins.

In contrast to the libc_pipe plugin, this plugin defers cross-handle
notification until I/O signal handling rather than block and unblock
readers using a semaphore. This is a performance regression in the case
of multiple threads blocking on a pipe, but shall be an intermediate
mechanism pending renovations within the libc VFS and threading layers.
As a side effect, threads blocked on a pipe might no [... cut]
commit fe47464c (2019-11-06): New VFS plugin for emulating POSIX pipes

Add a new plugin for creating pipes between pairs of VFS handles. It is
intended to replace the libc_pipe plugin, one of the last remaining libc
plugins.

In contrast to the libc_pipe plugin, this plugin defers cross-handle
notification until I/O signal handling rather than block and unblock
readers using a semaphore. This is a performance regression in the case
of multiple threads blocking on a pipe, but shall be an intermediate
mechanism pending renovations within the libc VFS and threading layers.
As a side effect, threads blocked on a pipe might no [... cut]
closed 2019-11-19 by chelmuth