genodelabs/genode #2306: vbox4: hangs sometimes during early boot

alex-ab, 2017-03-03:
On NOVA: win7_64 VMs are affected (independent of count of vCPUs)
On Muen/hw: win7_32 VM (single vCPU).

Part of the issue is, at least for the NOVA case, issue #2299 and #2230.

alex-ab, 2017-03-06:
@nfeske, @chelmuth: please consider 9f0e83f78aec15a9d9cd6bb372ecf06dbf18963d, 8615e47ebc9f894687a35b33d2ce0327df523b91 and 17921acddbca0cf6b9dd37b5bb960e58949b0db6 already for staging

chelmuth, 2017-03-15:
Please revisit this issue with current master containing all commits above.

alex-ab, 2017-03-15:
@reet-, @Kensan: may you please re-check for muen-hw. For nova it looks like before the release, so there still issues if multiple VMs concurrently are started and some of them don't manage to come up into a usable state (the vbox_auto_multiple case).

Kensan, 2017-03-15:
The status of **vbox_auto_win7** on base-hw/Muen remains the same: black screen after the initial VirtualBox boot screen is shown.

nfeske, 2017-12-13:
Is this issue still valid? Or may we close it?

alex-ab, 2017-12-14:
We may close it.

closed 2017-12-14 by alex-ab