genodelabs/genode #2307: trace_subject_reporter: show cpu_load per cpu via log

alex-ab, 2017-03-03:


alex-ab, 2017-03-06:
Added another version which is standalone (so - not part of trace_subject_reporter).

nfeske, 2017-03-06:
I like your last commit. ;-)

In my opinion, we may even call the component just `top` instead of `top_log` because according to our server naming scheme, `top_log` would wrongly suggest that the component provides a log service. @alex-ab what you you think?

alex-ab, 2017-03-06:
I'm fine, will change it.

alex-ab, 2017-03-06:
Done. Please check.

skalk, 2017-03-08:
@alex-ab is it necessary to run the trace test as SMP test? On qemu, we get horrible effects with SMP on hw because off spin-lock's during kernel initialization, which is no problem when using real hardware.

skalk, 2017-03-08:
and on the zynq platform we get this error: `Number of SMP cpus requested (2), exceeds max cpus supported by machine `xilinx-zynq-a9' (1)`

alex-ab, 2017-03-08:
smp would be good to test, but it would be sufficient for x86 or nova

skalk, 2017-03-08:
ok, thanks! I'll incorporate it.

skalk, 2017-03-08:
@nfeske @chelmuth : please consider commit 24fc11f for staging. It fixes all the currently broken trace tests triggered by the buildbot.

commit 0d10b7d7 (2017-03-07): os: app showing top style CPU utilization via LOG

Fixes #2307
closed 2017-03-15 by chelmuth