genodelabs/genode #2309: Cannot get NOVA demo to run on qemu

fidget324, 2017-03-07:
I followed the directions in the Genode book, Section 2.4, to build and run the nova demo on x86_64. I'm using Genode 17.02, and qemu 2.7.1 on Ubuntu 16.04 LTS. When I run
```
$ make run/demo
```
everything seems to build just fine, and then qemu launches. The GUI for the demo is then displayed, but freezes immediately. The scout window is displayed with the green x-ray shading, but I cannot move or manipulate it at all. The entire GUI just remains frozen, and I am unable to do anything. Here's the output when I run `make run/demo`:
```
including /home/scott/Documents/genode/tool/run/power_on/qemu
including /home/scott/Documents/genode/tool/run/log/qemu
including /home/scott/Documents/genode/tool/run/image/iso
including /home/scott/Documents/genode/tool/run/boot_dir/nova
including /home/scott/Documents/genode/repos/os/run/demo.run
building targets:  core init drivers/timer server/nitpicker app/pointer app/status_bar app/xray_trigger server/liquid_framebuffer app/launchpad app/scout test/nitpicker server/nitlog drivers/framebuffer drivers/input server/report_rom server/rom_filter drivers/platform drivers/acpi server/report_rom drivers/platform/spec/x86/device_pd
spawn make core init drivers/timer server/nitpicker app/pointer app/status_bar app/xray_trigger server/liquid_framebuffer app/launchpad app/scout test/nitpicker server/nitlog drivers/framebuffer drivers/input server/report_rom server/rom_filter drivers/platform drivers/acpi server/report_rom drivers/platform/spec/x86/device_pd
make[1]: Entering directory '/home/scott/Documents/genode/build/x86_64'
Makefile:134: target 'server/report_rom' given more than once in the same rule
checking library dependencies...
Skip target drivers/framebuffer/spec/exynos because it requires exynos
Skip target drivers/framebuffer/spec/imx53 because it requires imx53
Skip target drivers/framebuffer/spec/omap4 because it requires omap4
Skip target drivers/framebuffer/spec/pl11x/pbxa9 because it requires pl11x pbxa9
Skip target drivers/framebuffer/spec/rpi because it requires rpi
Skip target drivers/framebuffer/spec/sdl because it requires linux sdl
Skip target drivers/input/spec/imx53 because it requires imx53
Skip target drivers/input/spec/ps2/pl050 because it requires pl050
Skip target drivers/platform/spec/arndale because it requires arndale
Skip target drivers/platform/spec/imx53 because it requires imx53
Skip target drivers/platform/spec/odroid_x2 because it requires odroid_x2
Skip target drivers/platform/spec/rpi because it requires rpi
Skip target drivers/timer/fiasco because it requires fiasco
Skip target drivers/timer/foc because it requires foc
Skip target drivers/timer/hw because it requires hw
Skip target drivers/timer/linux because it requires linux
Skip target drivers/timer/pistachio because it requires pistachio
  Library ldso-startup
  Library blit
  Library scout_gfx
  Library mini_c
  Library startup-nova
  Library cxx
  Library x86emu
  Library alarm
  Library config
  Library ld
  Library libz_static
  Library timeout
  Library base-nova-common
  Library base
  Library libpng_static
  Program drivers/acpi/spec/x86/acpi_drv
  Program drivers/framebuffer/vesa/fb_drv
  Program drivers/input/dummy/dummy_input_drv
  Program drivers/input/spec/ps2/x86/ps2_drv
  Library core-nova
    COMPILE  version.o
    MERGE    core-nova.lib.a
  Library base-nova
  Program drivers/platform/spec/x86/platform_drv
  Program drivers/timer/nova/nova_timer_drv
  Program drivers/timer/pit/pit_timer_drv
  Program server/nitlog/nitlog
  Program init/init
  Program server/nitpicker/nitpicker
  Program server/rom_filter/rom_filter
  Program server/report_rom/report_rom
  Program test/nitpicker/testnit
  Library launchpad
  Program app/pointer/pointer
  Program app/status_bar/status_bar
  Library scout_widgets
  Program app/xray_trigger/xray_trigger
[... 163 lines cut ...]
[init -> acpi_drv] Found MADT
[init -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
[init -> acpi_drv] MADT IRQ 5 -> GSI 5 flags: 13
[init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13
[init -> acpi_drv] MADT IRQ 10 -> GSI 10 flags: 13
[init -> acpi_drv] MADT IRQ 11 -> GSI 11 flags: 13
[init -> acpi_drv] Warning: 2 dangling allocations at allocator destruction time
[init] child "platform_drv" announces service "Platform"
[init -> ps2_drv] Using keyboard with scan code set 1 (xlate)
[init -> ps2_drv] detected ExPS/2 mouse - activating scroll-wheel and 5-button support
[init -> platform_drv] PS2 uses IRQ, vector 0x1
[init -> platform_drv] PS2 uses IRQ, vector 0xc
[init] child "ps2_drv" announces service "Input"
[init -> fb_drv] Found PCI VGA at 00:02.0
[init -> fb_drv] fb mapped to 0x1000
[init] child "fb_drv" announces service "Framebuffer"
[init -> fb_drv] Found: VESA BIOS version 3.0
[init -> fb_drv] OEM: SeaBIOS VBE(C) 2011
[init -> fb_drv] Found: physical frame buffer at 0xfd000000 size: 16777216
[init -> fb_drv] fb mapped to 0xb000000
[init -> fb_drv] using video mode: 2560x1600@16
[init] child "nitpicker" announces service "Nitpicker"
[init -> nitpicker] Warning: not enough RAM to preserve buffer content during resize
[init -> scout] png is 101 x 92, depth=8
```

alex-ab, 2017-03-07:
When using the default qemu package as shipped by 16.04. Ubuntu LTS
```
QEMU emulator version 2.5.0 (Debian 1:2.5+dfsg-5ubuntu10.9), Copyright (c) 2003-2008 Fabrice Bellard
```

the demo works for me.

fidget324, 2017-03-07:
Hi Alexander,

Thanks for the quick reply. I installed the default emu package (2.5.0), but I'm still experiencing the exact same behavior. I'm wondering if it could be related to one of the warnings I'm getting:
```
[init -> nitpicker] Warning: not enough RAM to preserve buffer content during resize
```
I tried googling this warning, but did not see any results.

Scott

alex-ab, 2017-03-07:
You are sure 2.5.0 is active (not by accident still 2.7.1 is taken) ? What does
``` qemu-system-x86_64 --version```
tell you ?

In the meantime I downloaded and build 2.7.1 and yes, there it also does not work for me.

alex-ab, 2017-03-07:
Ahmm, you are running inside a VM or on native Ubuntu ?

fidget324, 2017-03-07:
I am definitely now using 2.5.0. I perhaps should have also mentioned that I'm using Ubuntu inside of VirtualBox. My host system is a 2016 MacBook Pro with MacOS Sierra.

alex-ab, 2017-03-07:
May you just for for run on the base-hw kernel instead of nova. For me base-hw work with 2.7.1

fidget324, 2017-03-07:
In etc/build.conf, I changed KERNEL from 'nova' to 'hw' and rebuilt the demo. When qemu launches, it now displays "Booting 'Genode on base-hw'", but then I run into the same issue: a frozen GUI. Although this time the warning messages from Genode are different:
```
using 'ld-hw.lib.so' as 'ld.lib.so'
using 'hw_timer_drv' as 'timer'
core link address is 0x10200000
install bootloader
creating ISO image...
Warning: -follow-links does not always work correctly; be careful.
Size of boot image is 4 sectors -> No emulation
 86.64% done, estimate finish Tue Mar  7 15:46:35 2017
Total translation table size: 2048
Total rockridge attributes bytes: 3281
Total directory bytes: 10968
Path table size(bytes): 64
Max brk space used 1e000
5771 extents written (11 MB)
spawn qemu-system-x86_64 -no-kvm -display sdl -m 256 -serial mon:stdio -cdrom var/run/demo.iso
warning: TCG doesn't support requested feature: CPUID.01H:ECX.vmx [bit 5]
Bender: Hello World.
Need 005e7000 bytes to relocate modules.
Relocating to 0f9f9000:
Copying 6189040 bytes...

kernel initialized
:virt_alloc: Allocator 0x1030f7f0 dump:
 Block: [0x1000,0x2000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x2000,0x3000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x3000,0x4000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x4000,0x5000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x5000,0x6000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x6000,0x7000] size=0x1000 avail=0x0 max_avail=0xf2000
 Block: [0x7000,0x8000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x8000,0x9000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x9000,0xa000] size=0x1000 avail=0x0 max_avail=0xf2000
 Block: [0xa000,0xb000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0xb000,0xc000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0xc000,0xd000] size=0x1000 avail=0x0 max_avail=0xf2000
 Block: [0xd000,0xe000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0xe000,0x100000] size=0xf2000 avail=0xf2000 max_avail=0xf2000
 Block: [0x101000,0x102000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x102000,0x103000] size=0x1000 avail=0x0 max_avail=0xcf7f2000
 Block: [0x103000,0x104000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x104000,0x200000] size=0xfc000 avail=0xfc000 max_avail=0xfc000
 Block: [0x9fc000,0x9fd000] size=0x1000 avail=0x0 max_avail=0xf803000
 Block: [0x9fd000,0x10200000] size=0xf803000 avail=0xf803000 max_avail=0xf803000
 Block: [0x1080d000,0x1080e000] size=0x1000 avail=0x0 max_avail=0xcf7f2000
 Block: [0x1080e000,0xe0000000] size=0xcf7f2000 avail=0xcf7f2000 max_avail=0xcf7f2000
 Block: [0xf0001000,0xf0002000] size=0x1000 avail=0x0 max_avail=0xcf7f2000
 Block: [0xf0003000,0xf0004000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0xf0005000,0xf0006000] size=0x1000 avail=0x0 max_avail=0xffe8000
 Block: [0xf0007000,0xfffef000] size=0xffe8000 avail=0xffe8000 max_avail=0xffe8000
 => mem_size=4011720704 (3825 MB) / mem_avail=4011634688 (3825 MB)

:phys_alloc: Allocator 0x1030e720 dump:
 Block: [0x1000,0x2000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x2000,0x3000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x3000,0x4000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x4000,0x5000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x5000,0x6000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x6000,0x7000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x7000,0x8000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x8000,0x9000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x9000,0xa000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0xa000,0xb000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0xb000,0xc000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0xc000,0xd000] size=0x1000 avail=0x0 max_avail=0xf5e4000
 Block: [0xd000,0xe000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0xe000,0xf000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0xf000,0x10000] size=0x1000 avail=0x0 max_avail=0x0
 Block: [0x10000,0x11000] size=0x1000 avail=0x0 max_avail=0x0
[... 63 lines cut ...]
[init -> acpi_drv] MADT IRQ 5 -> GSI 5 flags: 13
[init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13
[init -> acpi_drv] MADT IRQ 10 -> GSI 10 flags: 13
[init -> acpi_drv] MADT IRQ 11 -> GSI 11 flags: 13
[init -> acpi_drv] Warning: 2 dangling allocations at allocator destruction time
[init] child "platform_drv" announces service "Platform"
[init -> ps2_drv] Using keyboard with scan code set 1 (xlate)
[init -> ps2_drv] detected ExPS/2 mouse - activating scroll-wheel and 5-button support
[init -> platform_drv] PS2 uses IRQ, vector 0x1
[init -> platform_drv] PS2 uses IRQ, vector 0xc
[init -> fb_drv] Found PCI VGA at 00:02.0
[init -> fb_drv] fb mapped to 0x1000
[init] child "ps2_drv" announces service "Input"
[init] child "fb_drv" announces service "Framebuffer"
[init -> fb_drv] Found: VESA BIOS version 3.0
[init -> fb_drv] OEM: SeaBIOS VBE(C) 2011
[init -> fb_drv] Found: physical frame buffer at 0xfd000000 size: 16777216
[init -> fb_drv] fb mapped to 0xf0000000
[init -> fb_drv] using video mode: 2560x1600@16
[init] child "nitpicker" announces service "Nitpicker"
[init -> nitpicker] Warning: not enough RAM to preserve buffer content during resize
Warning: Quota exceeded! amount=130341, size=12288, consumed=122880
Warning: Quota exceeded! amount=154917, size=12288, consumed=147456
[init -> scout] png is 101 x 92, depth=8
```

alex-ab, 2017-03-07:
The output is ok, it should just work. Seems to be an issue with your virtualization stack. Since we have no Mac around here, we're not going to investigate.

What you may try is to move the generated .iso outside of your VM and run it as a own Virtualbox VM instance, maybe you have more luck this way. (Or you just dd the image to a usb stick and run it on a native machine, but then you probably have no serial output...).

alex-ab, 2017-03-07:
If you using the nova version of the demo, you have to enable the I/O APIC in the Virtualbox VM configuration !

cproc, 2017-03-07:
Does ```export SDL_VIDEO_X11_DGAMOUSE=0``` in the Ubuntu VM help? This feature is known to cause problems with input events when running Qemu/SDL in a VM.

fidget324, 2017-03-07:
Yes, that last fix worked! `export SDL_VIDEO_X11_DGAMOUSE=0` solved the problem. Slightly unrelated problem: the display in qemu is massive: 2560x1600@16. Is it possible to shrink this down?

cproc, 2017-03-07:
The display size can be configured in the 'demo.run' script for the 'fb_drv' component:
```
<start name="fb_drv">
        <resource name="RAM" quantum="4M"/>
        <provides><service name="Framebuffer"/></provides>
        <config width="800" height="600"/>
</start>
```

fidget324, 2017-03-07:
Works beautifully. Thanks! Now closing this thread...

closed 2017-03-07 by fidget324