genodelabs/genode #2315: ram_fs: stops serving files if they become to large

alex-ab, 2017-03-09:
Running vbox with overlay in RAM it may happen after a while:

```
[init] child "ram_fs" requests resources: ram_quota=2101248
[init -> ram_fs] Error: overlay_win7.vdi: size limit 2147483648 reached
[init -> ram_fs] Error: overlay_win7.vdi: size limit 2147483648 reached
```

Actual there is enough RAM available by the client.

nfeske, 2017-12-14:
This is actually not a bug but the result of the current dimensioning of the nested chunk data structure in `src/server/ram_fs/file.h` that is used to store the file content. The ram_fs should still work fine for other files as it explicitly handles the limit.

The data structure is organized hierarchical like a page table. Right now the limit is 4096*128*64*64. In principle, we could increase the value, at least on 64-bit. I am wondering, what would be a reasonable upper limit and how should we dimension the different levels in the best way?

Note that we must not use too large values on 32 bit to prevent integer overflows. The distinction between 32 and 64 bit could be done using the C++ traits pattern.

nfeske, 2019-03-17:
Increasing the limit to 8 GiB on 64-bit machines seems sensible.

Commit https://github.com/genodelabs/genode/commit/095d2e7737e43eb5367e7bec41d9993ed29b42ec makes the built-in heuristics more explicit and applies the parameters also to the VFS.

commit 095d2e77 (2019-03-17): ram_fs: increase max file size on 64 bit to 8 GiB

Fixes #2315
closed 2019-03-19 by chelmuth