genodelabs/genode #2329: read_rtc: rtc not configured, returning 0

rite, 2017-03-14:
To reproduce, make run/qt5 (x86_64, KERNEL=nova) and play some rounds of tetrix :)
042dafc doesn't resolves the issue, but removes some unrelated clutter in the log.

ehmry, 2017-03-15:
You wil want to apply the fix at 72d59719409070684d267487e8c3e0485b80e3e3.

ehmry, 2017-03-15:
This is the same issue as *libc walks config twice to 'libc'* #2313.

rite, 2017-03-15:
I applied 72d5971 (and 042dafc) on master but read_rtc still returns 0. What am I missing?

chelmuth, 2017-03-15:
A look into *var/run/qt5/genode/config* reveals that the scenario misses the rtc_drv and the corresponding VFS/libc configuration.

alex-ab, 2017-03-15:
In the configuration of <dir name="dev"> at least a <rtc/> and additionally the rtc_drv must be loaded and build, repos/libports/run/qt5_drivers.inc seems to be a good place to add.

chelmuth, 2018-08-29:
Commit f347cb90 addressed the immediate issue by configuring all examples with a static RTC value and libc config. The rtc_drv is still not part of *repos/libports/run/qt5_drivers.inc*, though. Still closing this issue as fixed.

closed 2018-08-29 by chelmuth