genodelabs/genode #2335: LwIP VFS plugin

ehmry, 2017-03-15:


ehmry, 2017-03-24:
I have lwIP working with netty at 863b1744198d4c880c8aa09cc4d6138df2973ca8 and 1760300b9b3844541acf8125bf6f5f7edc4095c0. Not perfect, fails with the netperf test but we need better tests to find out what is going on, netperf is quite complicated.

The lwIP update at 863b1744198d4c880c8aa09cc4d6138df2973ca8 strips away the synchronous and socket layers from the library, which means running single threaded, so a lot of support code and the patches are no longer needed. To use native lwIP in a component, call ``Lwip::lwip_init(Genode::Env&) and instantiate the ``Lwip::Nic_netif``, which will hook all the signal handlers for driving packets thru the stack.

chelmuth, 2017-03-27:
@ehmry I did not look into the patch but am I right thinking that your approach removes the lwip libc plugin?

ehmry, 2017-03-27:
@chelmuth It would, but I neglected to do so until bfbb9b6f8a8774c3203b8a77d48b7042fd57b545.

ehmry, 2017-05-12:
Rebasing...

ehmry, 2017-05-12:
The lwip plugin is ready for staging, works with all netty test variants and netperf.

The netperf results show more throughput than lxip with less memory. I suspect this is because write will block (return ``ERR_WOULD_BLOCK``) when the TCP congestion occurs but will never block for outstanding acknowledgements to come back. I don't know if POSIX defines that behavior but I will look into it. The memory is unsurprising, I managed to get a zero-copy receive between the Nic buffer and the stack.

My branch is at ``issue2335``, the required commits are: 1719cdae8, 7e02583c9, b3158b209, 9b29432d0, f894f5278, 3f7625674, 621ca4f6a.

ehmry, 2017-09-06:
Rebasing my ``issue2335_lwip`` branch to staging. Performance isn't great so I will worry about retuning the TCP settings next.

ehmry, 2017-09-07:
@chelmuth my ``issue2335_lwip`` branch is ready for staging, I have Netperf working with the Lxip plugin at 40f963f2e941bccdf76562094c51b12ed3a5ea20, so for whatever state lwIP is at, there should be no regressions introduced in this branch, if removing the lwIP plugin can be considered an improvement.

The improvements that are need at lwIP are measuring and tuning TCP performance, and moving some of lwIPs buffer allocations from heap to pools. For tuning I'd like to look into a special version of the VFS plugin that produces performance reports and a tool that can process these natively or externally.

ehmry, 2017-09-18:
Patch 419c5233cba099bbba84d28d230c900ab5317771 adds missing lwIP headers, ea567d74e7c7a943b5c8ff06b4f90190a7618216 will block the creation of lwIP sockets until the Nic interface is configured, and 20dc714aba0bade13c56a78dfe2ad792dd57fb89 make libc block for ``connect`` file sync completion when connecting.

ehmry, 2017-09-18:
Rebasing to staging...

chelmuth, 2017-10-04:
Current staging 0c36e98e3ffc15266c3a574a7d832fae17f226ac has a regression with noux_net (it still depends on libc_lwip_nic_dhcp) - try run/noux_net_netcat. Also, I still get segfaults with lwip under base-linux, e.g., netperf_lwip, which breaks nic_dump and nic_router run scripts. Oh, nic_router.run also has a regression regarding http_srv.
```
Error: target 'test/lwip/http_srv_static' does not exist
Error: target 'test/lwip/http_clnt' does not exist
```
@m-stein did you mention a partial fix of the script this morning? I could merge if you push and reference this issue.

Fixups on the issues mentioned above should get priority as they stall staging.

ehmry, 2017-10-04:
I have a fix for lwip on linux at c78b1c0fc2b577b24dfd0cb11f221298a88b1d5e and a nic_router fix at 9db147070d073f67458a1d237105fd36d4a00b59.

chelmuth, 2017-10-19:
I removed the fixed label for now.

ehmry, 2017-10-19:
Good, I will wait to see how well Lxip works.

NobodyIII, 2017-12-02:
This still hasn't been merged into genodelabs/master, or even staging, as far as I can tell. What is the status and when can I expect that to happen? Using lxip instead of lwip greatly increases the TCB size, if I understand correctly.

chelmuth, 2017-12-02:
We are working on it using the topic-branch development model as described in the Genode Foundations book. Expect more activity on the issue in December. Regarding the TCB, the usage of a TCP/IP VFS plugin increases the TCB of the applications using this plugin only. See the book Chapter 1 Paragraph "Application-specific trusted computing base".

BTW, please don't mistake the staging branch as an arbitrary testing area. Commits entering staging were carefully selected to make it to master with some fixups soon. Only occasionally, we have to remove commits from staging again for some reasons. All the development happens on the topic branches and commits are reviewed resp. shaped before merged into staging.

ehmry, 2017-12-13:
@chelmuth I've updated my `issue2335_lwip` branch to reflect the current state of the plugin. The challenge is that on my branch the LwIP library is stripped down to only the asynchronous API so the `libc_lwip` plugin must go, and that mean a lot more components and run scenarios have to be adjusted than I anticipated. Some of the run scenarios are a bit difficult to untangle and understand, so I would appreciate it if I could get a sense of what the priorities are.

chelmuth, 2017-12-18:
Highest priority must get run scripts that are in autopilot.list. The rest can be adapted step-by-step IMO.

ehmry, 2017-12-22:
I have new patches on my `issue2335_lwip` branch, notably a patch to restore the order of configuration in lwipopts.h 52c78496d23c8e8a175e7dac36cfaed74b32830a and a patch 1d4f584f03025c6c54b2935503eeef60fd23ff4b to provide lwip with the memcmp symbol that was missing when used with Noux.

ehmry, 2018-02-23:
@chelmuth Can I get 3725404898ef75addcc1d2028c660fca65464a15 merged so that the libc is compatible with the current vfs_lxip and external vfs_lwip packages?

chelmuth, 2018-02-23:
@ehmry I tested and merged the commit.

ehmry, 2018-06-25:
I've rebased the LwIP plugin to stagning on my `lwip` branch.

ehmry, 2018-06-25:
It might not be so useful, I don't think the LwIP DHCP client picks up the DNS configuration.

chelmuth, 2018-06-26:
There is `LWIP_DHCP_PROVIDE_DNS_SERVERS` which depends on `LWIP_DNS` config.

chelmuth, 2018-06-26:
With this config and If there are one or more name servers provided in the DHCP offer, it calls `dns_setserver()` from *dhcp.c*. I don't know if this is the place to hook in or if there's an API though.

ehmry, 2018-07-04:
@chelmuth yes, there is `dns_getserver` for getting the DHCP nameserver. I have rebased my `lwip` with a fixup that adds a `nameserver` file and patches `run/libc_getaddrinfo` to use LwIP.

I would say the plugin is ready for staging, and the `lwip_legacy` users should be migrated.

ehmry, 2018-07-10:
@m-stein I've removed the use of lwip_legacy completely from `run/nic_router` on my `2335_LwIP_VFS_plugin` branch. Seems to work.

m-stein, 2018-07-10:
@ehmry So far, I finished adaption of nic_router.run, nic_dump.run, nic_bridge.run, lwip.run, test/lwip on my 2335_LwIP_VFS_plugin branch. I deleted the http_srv_static component as it is not necessary anymore. The NIC router test now needs much less RAM, caps and time on all platforms. I was able to enable all tests of the NIC router test permanently and all clients seem to be stable now :)

m-stein, 2018-07-11:
@ehmry I've updated my branch with the attempt to adapt noux_net_netcat:
22fc1f6 noux_net_netcat: attempt to apply vfs lwip plugin
64e8537 fixup "WiP! Merge remove noux/net, uplift noux/minimal" (merging)
9b2d11f WiP! Merge remove noux/net, uplift noux/minimal

ehmry, 2018-07-11:
@m-stein I am taking a look now, I will know what is happening soon.

ehmry, 2018-07-11:
It seems to me that the asychronous I/O between LwIP in the VFS server and the Noux child is not working, I see from audit and the syscall logging that the bind and listen calls don't complete until after the first connect packet arrives at the VFS server.

ehmry, 2018-07-18:
@cnuke please add 5e27b527895ed82c784819e7f240f3eb0aea7a85 to staging, this adds the LwIP VFS plugin and a fetchurl test to autopilot. It does not effect lwip_legacy or vfs_lxip.

ehmry, 2018-07-18:
@cnuke I have an additional patch for staging. 852f8e42de780126e9f67dd5c6da495fe0594d27. This is @m-stein patch for `libports/src/test/lwip`.

cnuke, 2018-07-18:
@ehmry done.

ehmry, 2018-07-19:
@cnuke I have a fixup for the autopilot tests at 742c30ec303b3f676a8bedf77bc249576ea4adae.

cnuke, 2018-07-19:
@ehmry merged, thanks.

ehmry, 2018-07-20:
@cnuke I have a commit for staging, 31b922728434dfffcb490f771c060f618e1725dc.

cnuke, 2018-07-20:
@ehmry done.

ehmry, 2018-07-27:
@cnuke I have a LwIP fixup at 13c60e905bb89d6167a3d2d5e0a238050e4ed6eb. This is related to socket peeking, which is not supported in the libc, but can be safely added to the plugins until then.

cnuke, 2018-07-27:
@ehmry okay, merged to staging.

ehmry, 2018-07-30:
@cnuke I have a fixup for the Lwip release candidate recipe at 1ee33168e839c4316ee039651f8b8195b136a75e, ready for staging.

chelmuth, 2018-08-02:
The recipe hash is not stable for vfs_lwip as the following sequence shows. Please have a look.
```
> ./tool/depot/create UPDATE_VERSIONS=1 FORCE=1 genodelabs/src/vfs_lwip
created genodelabs/api/base/2018-07-12
created genodelabs/api/libc/2018-07-09
created genodelabs/api/nic_session/2018-05-28
created genodelabs/api/os/2018-05-28
created genodelabs/api/so/2017-04-24
created genodelabs/api/timer_session/2018-04-03
created genodelabs/api/vfs/2018-07-25
created genodelabs/src/vfs_lwip/2018-08-02 (new version)

> ./tool/depot/create UPDATE_VERSIONS=1 FORCE=1 genodelabs/src/vfs_lwip
created genodelabs/api/base/2018-07-12
created genodelabs/api/libc/2018-07-09
created genodelabs/api/nic_session/2018-05-28
created genodelabs/api/os/2018-05-28
created genodelabs/api/so/2017-04-24
created genodelabs/api/timer_session/2018-04-03
created genodelabs/api/vfs/2018-07-25
created genodelabs/src/vfs_lwip/2018-08-02-a (new version)

> ./tool/depot/create UPDATE_VERSIONS=1 FORCE=1 genodelabs/src/vfs_lwip
created genodelabs/api/base/2018-07-12
created genodelabs/api/libc/2018-07-09
created genodelabs/api/nic_session/2018-05-28
created genodelabs/api/os/2018-05-28
created genodelabs/api/so/2017-04-24
created genodelabs/api/timer_session/2018-04-03
created genodelabs/api/vfs/2018-07-25
created genodelabs/src/vfs_lwip/2018-08-02-b (new version)
```

chelmuth, 2018-08-02:
Merged with 40c231c8f459a6c8606b1ce15e21b2b880df6482

cproc, 2018-08-02:
I added run scripts for using the libc connect() test of issue #2920 with the lwip VFS plugin in [1]. It looks like the test does not return with lwip.

[1] https://github.com/cproc/genode/commit/e727b64e10dba808fe6724357b034a9fb0e5f93b

ehmry, 2018-08-07:
Connect error propagation fixed at c3fac5d746ffe43836f33408e50974c662052994.

closed 2018-08-08 by nfeske

nfeske, 2018-08-08:
The commit c3fac5d746ffe43836f33408e50974c662052994 does not apply to staging. @ehmry Am I missing another intermediate commit, or if not, could you please rebase the commit to the current version?

ehmry, 2018-08-08:
@nfeske try f02fe28813d0718c4855596ced8d670a51fc1266.

nfeske, 2018-08-08:
Thanks. Merged to staging.

ehmry, 2018-08-08:
I have one more little patch, 16e92619877650103325ca2b9043bd4465a89604. This shows a performance improvement in a synthetic test between two thinkpads over ethernet.

chelmuth, 2018-08-08:
Merged with d54175d7bca085589621786aded0a4098dddaa78.

ehmry, 2018-08-13:
Please consider 710477fa782c35c36d0cb116f218e04d2a7c46a0 for staging, it concerns defering connections server-side.

ehmry, 2018-08-13:
I have another commit 252c83df513ea862381b3a1cd100e565aadc5d5f, this one allows queued data to be read even if the connection is in an error state. This is the case when a one side of a connection is aborted but data is still queued for the application to process.

ehmry, 2018-08-24:
Please consider 77d1734ac9b4b77edab2de9a1b06fe0fd6cd79f1 for staging, it removes warning messages that are not warnings, and do not need to be logged.

commit d5cf7ce4 (2017-09-28): LwIP VFS plugin

This patch reintroduces the LwIP stack to libc as a VFS plugin implementing the socket_fs interface. Rather
than use LwIP's socket emulation layer this plugin interfaces directly to LwIP raw API and is single threaded.

The internal TCP parameters of the stack are untuned.

Fix #2335

fixup lib/vfs/lwip: queue ops for consistency
commit fc5870fd (2017-09-28): LwIP VFS plugin

This patch reintroduces the LwIP stack to libc as a VFS plugin implementing the socket_fs interface. Rather
than use LwIP's socket emulation layer this plugin interfaces directly to LwIP raw API and is single threaded.

The internal TCP parameters of the stack are untuned.

Fix #2335

fixup lib/vfs/lwip: queue ops for consistency
commit 9902350d (2017-12-20): LwIP VFS plugin

This patch reintroduces the LwIP stack to libc as a VFS plugin
implementing the socket_fs interface. Rather than use LwIP's socket
emulation layer this plugin interfaces directly to LwIP raw API and is
single threaded.

The internal TCP parameters of the stack are untuned.

Fix #2335
commit cf0d929f (2018-08-02): LwIP VFS plugin

This patch reintroduces the LwIP stack to libc as a VFS plugin
implementing the socket_fs interface. Rather than use LwIP's socket
emulation layer this plugin interfaces directly to LwIP raw API and is
single threaded.

The internal TCP parameters of the stack are untuned.

Fix #2050
Fix #2335