genodelabs/genode #2338: vbox5: enable all features available also in our vbox4 port

alex-ab, 2017-03-15:


chelmuth, 2017-03-16:
@alex-ab feel free to notify me if commits from this developments should be merged into staging early (resp. before the final issue is fixed).

alex-ab, 2017-03-16:
@chelmuth: I cleaned up and squashed some commits, so now you may add them to staging. Feature wise SMP support got added.

chelmuth, 2017-03-16:
Merge all commit incl. 8f49fe4 resp. 7824b0b

alex-ab, 2017-04-03:
09860a6457efb13c4829f094a4a34862d76829b0 (+ the fixes of #2362) enables the xhci model for vbox5

alex-ab, 2017-04-20:
shared folder test now also succeeds for vbox5

alex-ab, 2017-04-26:
The guest additions version 5* are now working also properly. Beforehand solely vbox guest addition of version 4* in vbox 5 were actually running fine.

closed 2017-06-01 by alex-ab