genodelabs/genode #2339: Tool for creating API, source, and binary archives

nfeske, 2017-03-15:
For realizing the package management for Genode (#1082), we need tools for extracting meaningful subsets of the source code from the source tree, and for transforming those pieces into working binaries.

The first question is what those "subsets" are. After intensive experimentation, I came up with the following distinction:

* An API archive contains the interface of a library. This interface may consist of header files, an ABI, build-description files (i.e., import.mk files), and even source code (if the library is a static library).
* A source archive contains the source code of either an executable binary or a shared library that implements the interface of a given API archive. A source archive may depend on several API archives.

GIven one source archive and all the API archives that are used by the source archive, a binary archive can be created. Given a specific target architecture, there is a one-to-one relationship between source archives and binary archives.

The new `depot` tool on the topic branch https://github.com/nfeske/genode/commits/pkg implements this idea.

I am speaking of archives - as opposed to packages - to highlight the fact that the archives are pretty dumb. Even though source archives carry their build-time dependencies (the used APIs) as information, there is no notion of runtime dependencies. The latter will be the job for another "layer" that will introduce the notion of "packages" as a collection of binary archives.

The `depot` tool comes in the following three parts, which operate on the new `<genode-dir>/depot/` directory. This directory is created on demand.

* `tool/depot/api` creates an API archive for a given API name. The content of the archive is defined by a so-called recipe that can be located at `<repo>/recipes/api/<api-name>/` of any source-code repository. API archives are created at `depot/api/<api-name>-<api-version>`.
* `tool/depot/src` creates a source archive based in a recipe located at `<repo>/recipe/src/<archive-name>/`. Source archives are created at `depot/src/<src-name>-<src-version>`.
* `tool/depot/bin` creates a binary archive for the specified source-archive name. Because the binary is specific for a certain platform, the platform must be specified in the form of a SPEC value, for example `SPEC=x86_32`. Binary archives are created at `depot/bin/<SPEC>/`. Regular executables are placed at the respective sub directory `<src-name>-<src-version>`. If the source archive is a library (if it implements an API), it is placed at the sub directory `<api-name>-<api-version>/<src-name>-<src-version>`. This way, multiple implementations of the same API (e.g., bugfixed minor versions) can peacefully coexist.

Version identifiers are maintained manually but with the assistance of a content hash. Originally, I planned to follow the Nix approach of using content hashes as version identifiers. But given the nested structure of the naming of binary archives introduced by the notion of API packages, using hashes as version identifiers becomes rather cryptic. Another advantage of human definable versions is the ability to order versions. E.g., by using a release date as version number, we intuitively have the tangible notion of the "newest" version, which is much easier to grasp and handle than some piece of meta data stored inside a hash-named directory.

That said, content hashes are still used as a tool to aid the versioning of archives. In fact, each recipe is accompanied with a `hash` file. This file contains both the (human-managed) version identifier and the hash of the archive's content. Whenever an API or source archive is created, the `depot` tool compares the hash found in the recipe's hash file with the hash of the actual content. If they differ, the tool backs out with the instructions of how to update the recipe's hash. The given message proposes the use of the current date as version identifier, but the choice of the identifier is ultimately up to the maintainer.

Each recipe comes in the form of a directory that contains a `content.mk` file (defining the archive's content via simple make rules), a `hash` file. Source recipes can furthermore have a `used_apis` file, which denotes the list of the used API archives, as well as an `api` file that declares the name of the API implemented by the source archive. The latter is only present for libraries. In fact, the existence of the `api` file distinguishes library archives from executable archives. The `depot` tool is implemented via plain make. So it does not introduce a new tool dependency or a new language.

At the topic branch, I added a few recipes, which allow the building of zlib and report_rom as two examples.

nfeske, 2017-03-17:
After further discussion with @chelmuth, we came up with a nice idea for simplifying the front end of the depot tool, which I implemented in the latest version of the topic branch cited above.

Instead of having distinct tools for the different kinds of archives, we use the convention of how the depot is structured. There is a single 'create' tool, which accepts a path relative to the root of the depot. Given the convention of the depot's directory hierarchy, the different path elements implicitly define the type of the archive, the SPEC of binary archives, the name of the recipe, or - that is new - the identity of the package creator.

For example, if the user "alan" wants to populate his depot with a number of archives, he uses the `tool/depot/create` tool as follows:

* _./tool/depot/create alan/api/libc_ 
  Creates an API archive for the libc, which will be located at `depot/alan/api/libc-<version>`.
* _./tool/depot/create alan/src/zlib_
  Creates a source archive for zlib library, which will be located at `depot/alan/src/zlib-<api-version>/zlib-<lib-version>`.
* _./tool/depot/create alan/src/report_rom_
  Creates a source archive for the report-rom component, which will be located at `depot/alan/src/report_rom-<version>`.
* _./tool/depot/create alan/bin/x86_64/report_rom_
  Creates a binary archive of the report-rom component for the x86_64 architecture. It will be located
  at `depot/alan/bin/x86_64/report_rom-<version>`.

The introduction of the identity of the package creator has the following background:

* The identity will be used as the basis for signing and verifying archives. The presence of this identity at the the first-level subdirectory in the depot gives us a natural place for storing information specific for the identity, i.e., the package creator's respective public key. So the depot can be used as a key ring (for public keys) in a very natural way.
* It will eventually be possible to import archives of other users into the depot (similar how git remotes can be used to import foreign content into a git repository). Each "remote" lives nicely in its own subdirectory. So the origin of archives is always clear.
* The identity as first-level path element scopes the name space of archive versions. E.g., if two packagers create new versions of a package simultaneously with conflicting version numbers, this is no problem because each packager publishes his archive under his own identity (proved by the package creator's signature). When retrieving archives, the user refers to an archive with the packager's identity, the package name, and the packager-defined version.

nfeske, 2017-03-21:
I just updated my topic branch with a new version of the depot tool that supports the notion of pkg archives. Since the internal design and implementation of the tool is still in flux, I squashed all intermediate states into a single commit.

Regarding "packages", a pkg recipe corresponds to a collection of archives given in the file `<repo>/recipes/pkg/<pkg-name>/archives`. Each line corresponds to a _src_ or _pkg_ archive as stored in the depot. E.g., a `pkg/desktop/archives` file may look as follows:

```
_/src/report_rom
_/pkg/panel
skalk/pkg/backdrop
```

The lines with the placeholder `_` as user refer to a local archive as generated from the Genode source tree. The user corresponds to whichever user is specified to the `create` tool. That is, in the generated `pkg` archive within the depot, the `_` will be replaced by the concrete user. The third line refers to a "foreign" archive, which does not originate from the local Genode repository but is imported into the depot via another way (e.g., downloaded packages). The depot/create tool ignored those lines.

Each pkg-recipe directory must contain a README file explaining the purpose of the package.

The recipe directory may contain any number of further files, which are copied as is into the generated archive. Note that those files are not meant as actual input to the packaged Genode scenario but are considered as meta data. E.g., for a Genode subsystem delivered as a pkg archive, the archive may contain the information about the subsystem's expected execution environment.

The `archives` file does not list any `api` or `bin` archives. The information about the needed `api` archives is stored in the API-using `src` archives. Any `bin` archive corresponds to one `src` archive. So by specifying the `src` archive, the name of the corresponding binary archive can be inferred automatically.

The _create_ tool is able to determine all dependencies of a given `pkg` and create all dependent archives automatically. E.g., `./tool/depot/create alan/pkg/desktop` will create all api, src, and pkg archives needed for the "desktop". When going to the binary level, the same usage pattern applies. By issuing `./tool/depot/create alan/pkg/x86_64/desktop`, all api, src, pkg, and bin archives are created that are needed for the "desktop" on the x86_64 target platform.

nfeske, 2017-03-27:
In the new version of the topic branch, I supplemented the create tool with two new tools: _publish_ and _download_.

As the name suggests, the _publish_ tool transforms depot content into a publishable form. That is, it creates compressed and signed tar archives. The result is written to `<genode-dir>/public/`. For a given argument (there can actually be more than one), the tool determines all dependencies automatically and "publishes" everything that is needed to satisfy the needs of the argument. For signing the result, the tool takes the public key of the archive's user as stored at `<depot-dir>/<user>/pubkey` as a selector for the to-be-used private key (which is assumed to be present in the user's gpg key ring). This way, one user can publish depot content under various identities.

The _download_ tool performs the inverse operation. Given an archive path (or multiple), it iteratively downloads archives and determines transitive dependencies. The archives are downloaded to `<genode-dir>/public/`, verified against the public key as stored in the archive's respective `<depot>/<user>/pubkey` file, and unpacked into the depot. As of now, the download tool uses a dummy backend (mk/cp_downloader) that merely copies files from the directory `<genode-dir>/remote`. This is just for testing. A curl-based back end will follow.

nfeske, 2017-03-28:
In the latest version of my topic branch, I added support for raw-data archives. Those archives are designated for any content that is not created from a source archive and is architecture-independent. The recipes of raw archives are located at `<repo>/recipes/raw/<archive-name>/`. Each recipe contains a `hash` and `content.mk` file, just like the recipes for source archives. Raw-data archives can solely appear at the `archives` list of pkg archives.

ehmry, 2017-04-03:
If I may make a suggestion for one direction of distribution tooling, I think it is possible and desirable to make offline updates possible. If the depots were published as a sequence of signed transactions that reference content-addressed blobs, then updates are transaction log replays and a blob store, which may be distributed by network or physical media. Only a small set of transaction types would need be defined, something like Add, Update, Release, Deprecate, and Revoke.

Git could of course do this, but this only requires a subset of git's features (and not the confusing ones).

nfeske, 2017-04-04:
@ehmry you raise an interesting point but I feel it is too early for a definite answer of how to approach offline updates using the depot structure. Let us first use the tools for a while to see how they fit.

Maybe, the problem is alleviated by the depot structure, which already separates content (api/source/binary/raw archives) from meta data (package archives)?

Or maybe, the feature could be realized atop the depot tooling (thinking of diff'ing the output of `depot/dependencies` of different versions)?

Or maybe, rsyncing depots would be the most practical way?

The right answer may also very much depend on the scale of the depots and the diversity of installations.

nfeske, 2017-04-21:
I merged the intermediate result of this line of work to staging.

For experimenting with the depot, I adjusted the interactive run scripts of the gems repository to use the depot instead of the build system (except for a very few components). So these run scripts are currently the best playground to get acquainted with the tools. The current state covers all kernels for x86_32, x86_64, as well as base-hw for PBX-A9 (as a representative of ARM platforms).

The usage information of the individual tools under `tool/depot/` is given when invoking them without argument.

nfeske, 2017-06-01:
Integrated in Genode 17.05

closed 2017-06-01 by nfeske

skalk, 2018-02-02:
@nfeske please consider commit d6dd84e as a fixup, without that option it happens under certain circumstances that "bin" is added to the list of modules, and thereby gets added to the boot_modules assembler file, and the assembler aborts when trying to allocate a lot of memory

skalk, 2018-02-05:
@nfeske after consulting @chelmuth I agreed to discard above fixup. We found the actual root inside of platform.inc (commit f141f024b2155a6f80ff3f36af5bc79fdb749040), so you can ignore above comment.

nfeske, 2018-02-06:
Thanks for the notice.