genodelabs/genode #2341: tool/tool_chain x86 fails to build with gcc 6.3.1

pkerling, 2017-03-16:
I get the following error when trying to build the x86 toolchain with `tool/tool_chain x86`:
```
In file included from /tmp/genode/contrib/gcc-4.9.2/gcc/cp/except.c:1013:0:
cfns.gperf: In function 'const char* libc_name_p(const char*, unsigned int)':
cfns.gperf:101:1: error: 'const char* libc_name_p(const char*, unsigned int)' redeclared inline with 'gnu_inline' attribute
cfns.gperf:26:14: note: 'const char* libc_name_p(const char*, unsigned int)' previously declared here
cfns.gperf: At global scope:
cfns.gperf:26:14: warning: inline function 'const char* libc_name_p(const char*, unsigned int)' used but never defined
make[3]: *** [Makefile:1058: cp/except.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:3967: all-gcc] Error 2
make[1]: *** [Makefile:856: all] Error 2
make: *** [tool/tool_chain:435: build/x86/gcc/gcc/g++] Error 2
```

This seems to be caused by an attribute mismatch between definition and declaration of `libc_name_p` that was previously ignored by the compiler but now causes an error, at least with gcc 6.3.1.

Output of `gcc -v`:
```
Using built-in specs.
COLLECT_GCC=/usr/bin/gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/6.3.1/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc-multilib/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++ --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --enable-multilib --disable-werror --enable-checking=release
Thread model: posix
gcc version 6.3.1 20170306 (GCC) 
```

pkerling, 2017-03-16:
Related: https://gcc.gnu.org/ml/gcc-patches/2015-08/msg00375.html

nfeske, 2017-03-17:
@pkerling Thank you very much for the patch, which I'd like to merge as is. @cproc Do you approve?

cproc, 2017-03-17:
@nfeske: I'm fine with it, but the GCC port hash needs to be updated.

pkerling, 2017-03-17:
I ran `tool/ports/update_hash gcc`, did that fix it?

nfeske, 2017-03-18:
Yes, that's it. Thank you for updating the commit so quickly. I merged it to the genodelabs/staging branch just now.

As a minor cosmetic change, I moved the "fix #2342" part from the first line of the commit message to the body to stay within the commonly recommended line length of 50 characters for commit titles. I hope you are fine with this change.

Thank you for the contribution!

pkerling, 2017-03-18:
Sure, that's fine - closing the issue.

commit 8ecc9e65 (2017-03-18): Fix toolchain gcc compilation with newer host gcc

An inline attribute mismatch in the gcc-4.9.2 source causes its
compilation to fail on modern gcc versions (verified with gcc 6.3.1,
but probably since gcc 5)
The patch is courtesy of the gcc-patches mailing list:
https://gcc.gnu.org/ml/gcc-patches/2015-08/msg00375.html

Fixes #2341
closed 2017-03-18 by pkerling