genodelabs/genode #2343: acpica: fix: Uncaught exception of type 'Platform::Session::Out_of_metadata'

MartijnVerschoor, 2017-03-17:
After updating a scenario to 17.02 that includes acpica I get the following error:

[init -> acpica] detected - ec
[init -> acpica] Error: Uncaught exception of type 'Platform::Session::Out_of_metadata'
[init -> acpica] Warning: abort called - thread: ep

The acpica.run test script gives the same error (log attached). I tried updating the x86 platform session quota to 64K, but this did not help.

Any clues?

[acpica_log.txt](https://github.com/genodelabs/genode/files/851114/acpica_log.txt)

alex-ab, 2017-03-17:
Please try following commit, it solves it for me on a X201.

MartijnVerschoor, 2017-03-20:
This solves the issue for the X250 too.

Thanks! I'll close the issue.

commit 1d3dd900 (2017-03-20): acpica: increase memory quota for platform session

Fixes #2343
closed 2017-03-20 by MartijnVerschoor