genodelabs/genode #2347: hw core: generic timer driver interface

m-stein, 2017-03-23:


m-stein, 2017-03-23:
@skalk Could you please have a look at these commits. I had some spare time these days and this is what happened ^^ I don't know if they clash with your current base-hw development. If they do, we can also delay them.
0a0301b hw core: generic timer driver interface
f626ea4 hw cortex-a9 timer: raise translation precision

skalk, 2017-03-27:
@m-stein: indeed it would be really nice if we delay the re-structuring commit. The Cortex A9 timer fix of course can be applied immediately of course

chelmuth, 2017-03-28:
Merged 481660e89e197b766ed79e1a545c99b92fec0ef0 just now.

m-stein, 2017-03-30:
Solved a bug in the new kernel timer driver:
8385f1d fixup "hw cortex-a9 timer: raise translation precision"

m-stein, 2017-05-13:
This commit
29ab328 hw core: merge Kernel::Clock and Kernel::Timer
is the rebased version of the original fix that adds further simplification and a good diff-stat-karma.

m-stein, 2017-05-13:
@chelmuth A little clean-up:
ea1710c base-hw: Rename 'tics' 'ticks'

m-stein, 2017-05-14:
@chelmuth There's a fixup for muen
ad1cd93 Fixup "hw core: merge Kernel::Clock and Kernel::Timer"

commit 583e3e89 (2017-05-13): hw core: merge Kernel::Clock and Kernel::Timer

With this, we get rid of platform specific timer interfaces. The new
Timer class does the same as the old Clock class and has a generic
interface. The old Timer class was merely used by the old Clock class.
Also, we get rid of having only one timer instance which we tell with
each method call for which CPU it shall be done. Instead now each Cpu
object has its own Timer member that knows the CPU it works for.

Fixes #2347
closed 2017-05-31 by chelmuth