genodelabs/genode #2350: slab: total_counter gets wrong when already free slabs are freed again (double++ free)

alex-ab, 2017-03-23:
This is obviously a programming bug.

Unfortunately the slab total_counter gets confused and reports available free slots. If then a allocation is tried on the confused slab and the slab is close to empty, it will not try to expand with a new slab blocks but instead will return a nullpointer. This may cause, e.g., in qt5 to throw a std::bad_alloc exception.

Unfortunately this corrupted state will be detected partially long long long time later (after several allocations/frees) in code paths which are in principle absolutely ok, but they get a exception because of wrong frees which happened much earlier.

chelmuth, 2017-03-24:
In dca8e38, I added a test case for this issue. Unfortunately, the test does work on 32bit only because of the assumed `HEAP_OVERHEAD`, which seems to differ between 32 and 64 bit. Nevertheless, I'd like to enable the test for the autopilot. @nfeske do you have an idea to make the test independent of the arch or should I just enable it for 32 bit?

alex-ab, 2017-03-31:
How we want to proceed here ?

nfeske, 2017-03-31:
I have no objection against merging @alex-ab's detection patch.

chelmuth, 2017-03-31:
May I enable the slab test in autopilot for x86_32 only? I don't know how to *calculate* `HEAP_OVERHEAD` safely.

chelmuth, 2017-03-31:
Okay, I did the most simple calculation and enabled the test for non-qemu in autopilot.

commit b6c7a70e (2017-03-31): slab: detect corrupted slab and invalid frees

and report about that.

Fixes #2350
commit 7cc11138 (2017-04-11): slab: detect corrupted slab and invalid frees

and report about that.

Fixes #2350
commit a0c449f6 (2017-05-02): slab: detect corrupted slab and invalid frees

and report about that.

Fixes #2350
commit 1c79ba41 (2017-05-02): slab: detect corrupted slab and invalid frees

and report about that.

Fixes #2350
closed 2017-05-31 by chelmuth