genodelabs/genode #2351: Malformed config throws Parent::Quota_exceeded

BorisMulder-CSL, 2017-03-24:
I had an error in my route config:

```
<route>
			<any-service>
				<service name="LOG"> <child name="logtee"/> </service>
				<child name="wm"/> <parent/> <any-child/>
			</any-service>
</route>
```

Of course the correct route would be
```
<route>
			<service name="LOG"> <child name="logtee"/> </service>
			<any-service>
				<child name="wm"/> <parent/> <any-child/>
			</any-service>
</route>
```

However, Genode gives a Parent::Quota_exceeded error on a session creation to the "wm" child instead of giving a useful error when running the component with the malformed route. It took me a while to find it, first I tried giving more quota to the components until I found the mistake manually. Maybe it makes sense to add better error handling to detect these kinds of mistakes?

chelmuth, 2017-03-27:
This issue comprises two aspects.
1. Why is `Parent::Quota_exceeded` thrown in this case?
  For further investigation it would be helpful to have more information about the scenario triggering the issue or even better a minimal run script for reproducing the error. My expectation would have been that init just ignores the `<service>` declaration and tries to route *logtee*'s session request to *wm*, the parent, or any child. Also, I can't derive from your explanation which component experiences the `Quota_exceeded`. I'm definitely interested in investigating further in this regard.
2. Should init *check* the configuration for validity?
  I know that debugging these issues is annoying at times and were also tempted myself to add more code to our components to guide developers in these cases. But in any event, it boils down to added complexity in the implementation, which must be taken into account for potential bugs/security holes. I hope there's no doubt that we can't add an XML validator to our base components. So, the rule of thumb currently is: If the diagnostic information arises from the existing robustness measures of the implementation, it should be presented to the developer. Adding significant amount of code is a matter of discussion (and prototype implementation).

nfeske, 2017-03-27:
I share @chelmuth's sentiment. In the longer term, we might address the second point by the means of offline-validation tooling. Right now, we already have the xml-lint syntax checker, which is of great help already. A more sophisticated checking tool could in principle look out for additional invariants like:

* Are all start node uniquely named?
* Are priorities configured such that all servers always have a higher priority than their clients?
* Is the XML structure correct (detecting @BorisMulder-CSL's mistake)?
* Absence of wildcard rules (`<any-service>`, `<any-child>`, `<default-route>`, `<default-policy>`) whenever scenarios are built for deployment, not just development.

By embedding knowledge about components into the checker, the tool could go even further:
* Do the policy-selection rules of report_rom match the actual children of init (detecting spelling errors in those rules)?
* Can the prerequisites of components actually be satisfied? E.g., Is there a route from a file-system server to a block server? Is there a route from part_blk to a block server? Do the ROM modules configured in a VFS configuration actually exist? Is there a route from a device driver to the platform driver? Is there a platform-driver policy for the driver? Are there ambiguous routes? Are there superfluous routes that can never be taken?

The advantage of implementing such assistance features as an offline tool is that init's complexity remains unaffected. That said, we have no plans to build such a tool in the near term.

BorisMulder-CSL, 2017-12-14:
I could not quickly find the old scenario to debug this problem or provide more details on it, nor have I experienced this error again meanwhile. I will close this issue for now.

closed 2017-12-14 by BorisMulder-CSL