genodelabs/genode #2352: Background backdrop not displaying

BorisMulder-CSL, 2017-03-24:
When I create a backdrop that is not inside a window (i.e. it is routed to nitpicker directly instead of through a window manager), it does not display correctly.

You can verify this with gems/run/wm.run; for me it does not display the grid background.

nfeske, 2017-03-24:
The problem is related to the changed way of how sessions are prefixed in version 16.11 (http://genode.org/documentation/release-notes/16.11#Unification_and_tightening_of_session_labels). Please see the commit https://github.com/genodelabs/genode/commit/9bceec4394394401766fe689d617951ff4f784d2 for an update of the wm.run and nano3d.run scripts.

Nitpicker has a special background layer that sits behind all other views. A client can express that a certain view is its background view. E.g., a seamless window system like rdesktop would do that for the desktop background. Whenever the corresponding session has the user focus, the session's background view is shown, but never if any other session has the focus. This way, the background of multiple concurrently running window systems are nicely switched along with the change of focus. However, this only works as long as all participants follow the same protocol by marking their background views as such. If one client fails to do so and just presents a fullscreen view behind all others, no background view of the other sessions would ever be visible. For this reason, the backdrop program plays nice by marking its own view as a background.

Now, the backdrop program itself never gets the focus. Consequently, its background view would never have the chance to become visible. In order to enable the backdrop to be the provider of the default background that is shown when no session-specific background is displayed, nitpicker assigns the special role of the default-background provider to the client labeled as "backdrop".

The last part went wrong now. Since we changed the label prefixing, the backdrop client would no longer show up as "backdrop" but as "backdrop -> " at nitpicker. Furthermore, in commit 8ab932a8d09a0b9ed592322c5d335b4a83a83385, the label was changed to "backdrop -> backdrop". Consequently, nitpicker would never assign the default-background role to backdrop.

The immediate fix as done by the commit above is to explicitly set the label to the string expected by nitpicker.

But that said, I think that a better longer-term fix would be to manage backgrounds per domain, not per session. This way, any client of a certain domain could provide a background that will be shown whenever any session of this domain has the focus. This way, we could remove this magic "backdrop" label handling from nitpicker. But that is for another day.

nfeske, 2017-06-01:
Fixed in master.

closed 2017-06-01 by nfeske